volkszaehler/volkszaehler.org #828: aggregate fails with  SQLSTATE[HY000]: General error: 1206 The total number of locks exceeds the      lock table size

WolfgangFahl, 2020-10-07:
setup is the same as in #827 

```
/usr/bin/php /srv/www/volkszaehler.org.2020/bin/aggregate run -l day
Performing 'delta' aggregation on 'day' level

 [>---------------------------]   0%  < 1 sec/< 1 sec   0 channels
 [==>-------------------------]  10%  23 secs/3 mins    1 channels
 [=====>----------------------]  20%   3 mins/19 mins   2 channels
In AbstractMySQLDriver.php line 128:
                                                                               
  An exception occurred while executing 'REPLACE INTO aggregate (channel_id,   
  type, timestamp, value, count) SELECT channel_id, ? AS type, MAX(agg.timest  
  amp) AS timestamp, COALESCE( SUM(agg.val_by_time) / (MAX(agg.timestamp) - M  
  IN(agg.prev_timestamp)), AVG(agg.value)) AS value, COUNT(agg.value) AS coun  
  t FROM ( SELECT channel_id, timestamp, value, value * (timestamp - @prev_ti  
  mestamp) AS val_by_time, COALESCE(@prev_timestamp, 0) AS prev_timestamp, @p  
  rev_timestamp := timestamp FROM data CROSS JOIN (SELECT @prev_timestamp :=   
  UNIX_TIMESTAMP(DATE_ADD(FROM_UNIXTIME(MAX(timestamp) / 1000, "%Y-%m-%d"), I  
  NTERVAL 1 day)) * 1000 FROM aggregate WHERE type = ? AND aggregate.channel_  
  id = ?) AS vars WHERE channel_id = ? AND timestamp >= IFNULL((SELECT UNIX_T  
  IMESTAMP(DATE_ADD(FROM_UNIXTIME(MAX(timestamp) / 1000, "%Y-%m-%d"), INTERVA  
  L 1 day)) * 1000 FROM aggregate WHERE type = ? AND aggregate.channel_id = ?  
   ), 0) AND timestamp < UNIX_TIMESTAMP(DATE_FORMAT(NOW(), "%Y-%m-%d")) * 100  
  0 ) AS agg GROUP BY channel_id, YEAR(FROM_UNIXTIME(timestamp/1000)), DAYOFY  
  EAR(FROM_UNIXTIME(timestamp/1000))' with params [3, 3, "3", "3", 3, "3"]:    
                                                                               
  SQLSTATE[HY000]: General error: 1206 The total number of locks exceeds the   
  lock table size                                                              
                                                                               

In Exception.php line 18:
                                                                               
  SQLSTATE[HY000]: General error: 1206 The total number of locks exceeds the   
  lock table size                                                              
                                                                               

In PDOStatement.php line 141:
                                                                               
  SQLSTATE[HY000]: General error: 1206 The total number of locks exceeds the   
  lock table size                                                              
                                                                               

run [-l|--level LEVEL] [-m|--mode MODE] [-p|--periods PERIODS] [-v|--verbose] [--] [<uuid>...]

```

WolfgangFahl, 2020-10-07:
see e.g. https://stackoverflow.com/questions/6901108/the-total-number-of-locks-exceeds-the-lock-table-size

WolfgangFahl, 2020-10-07:
I am trying:
```sql
mysql> SET GLOBAL innodb_buffer_pool_size=402653184;
Query OK, 0 rows affected (0.00 sec)

mysql>  SELECT @@innodb_buffer_pool_size/1024/1024/1024;
+------------------------------------------+
| @@innodb_buffer_pool_size/1024/1024/1024 |
+------------------------------------------+
|                           0.375000000000 |
+------------------------------------------+
1 row in set (0.00 sec)
```
and restarting the aggregate

andig, 2020-10-07:
Ich vermute du schreibst gleichzeitig hochfrequent in die Tabelle? Dann würde ich das schreiben abstellen solange die initale Aggregation läuft.

WolfgangFahl, 2020-10-08:
@andig Danke für den Hinweis. Ja während des Aggregate kommen Daten an. Ich brauch das daily Aggregate nicht wirklich, weil sbfspot das auch schon kann. Das monthly scheint zu laufen und steht im crontab - mal gucken ob es da in den nächsten Monaten Probleme gibt.