volkszaehler/volkszaehler.org #834: Performance issue with Raspberry Pi Zero known?

xyNNN, 2020-12-04:
Hi!

The application is running on a Raspberry PI zero and works very well, thanks for this awesome product! 
Are there any known issues to run this on a Raspberry Pi Zero? I've encountered the problem that some values couldn't be logged, the graph in the backend is empty between some date/time ranges. I can't see any problems, on the webserver level I can see all the requests in the access logs, no issues in the error log ... and no other problems. Looks like the database is perhaps overwhelmed with the requests - but I don't know.

On a normal Raspberry Pi this application works without any outtakes.

Any suggestions or hints to identify the problem?

Thanks!

J-A-U, 2020-12-04:
> Any suggestions or hints to identify the problem?

Use more potent hardware. Like a full size Rpi. The zero is actually downsized to reduce dissipation and get it smaller.
The Image we provide "works" on a zero and is okay to have a ready to use vzlogger and feed another Rpi or even better hardware with. I don't recommend middleware and database on them.

xyNNN, 2020-12-05:
Thanks! I thought about it but I couldn't imagine that this is the problem. I will use a normal Rpi to use middleware and database.

Thanks!

closed 2020-12-05 by xyNNN

DanielGlaas, 2021-02-10:
Hi, I had the same problems with the Pi zero.
My solution is to enabled the watchdog such that it restarts the pi if no updated data is written into the log.

1. Enable the watchdog as described [here](https://diode.io/raspberry%20pi/running-forever-with-the-raspberry-pi-hardware-watchdog-20202/)
2. Configure "verbosity" to 5 in vzlogger.conf
> {
    // General settings
    "daemon": true,         // run periodically
    "verbosity": 5,         // log verbosity (0=log_alert, 1=log_error, 3=log_warning, 5=log_info, 10=log_debug, 15=log_finest)
    "log": "/home/pi/vzlogger.log", // log file, optional

3. sudo nano /etc/watchdog.conf, add
>  file                    = /home/pi/vzlogger.log
change                  = 10
interval		= 3
logtick                 = 10

This lets the watchdog check the log for changes every three seconds. Every 10th interval (=30 seconds), the watchdog prints a status info what it is doing

4. sudo nano /etc/default/watchdog, add
>watchdog_options="-v"

With this additional line you get the following output:

>sudo systemctl status watchdog.service 
Feb 10 19:54:24 volkszaehler watchdog[1039]: file /home/pi/vzlogger.log was last changed at Wed Feb 1
Feb 10 19:54:27 volkszaehler watchdog[1039]: still alive after 76460 interval(s)
Feb 10 19:54:54 volkszaehler watchdog[1039]: file /home/pi/vzlogger.log was last changed at Wed Feb 1
Feb 10 19:54:57 volkszaehler watchdog[1039]: still alive after 76470 interval(s)

This approach runs very stable with the ready-to-use image on Raspberry Pi Zero W.