volkszaehler/volkszaehler.org #836: Just running push-server and middleware but the total values of the meters are not in sync with last vzlogger push

bitstreamout, 2020-12-16:
Observered after two days.  I have beside others three channels for Bezug, Einspeisung, and Erzeugung and are all three electric meters and with correct initial values and scale factor 1000 set for the electric power here.  The first two channels are done with sml and the third with exec protocol.

Comparing the vzlogger log file with the last entries shows that the Gesamt/total consumptions are not in sync with those shown by the middleware server. Currently shown for Bezug/chn0 is 427 kWh, but actual value is 432769.20 W, for Einspeisung/chn1 it is 61.0 kWh, but value is  61140.30 W (that is beside the missing rounding for initial value OK ... not enough sun for Einspeisung only for Eigenverbrauch), and then for Erzeugung/chn4 it is 458 kWh, but actual value is 460057.00 W

The resulting curves  seems to be correct and are perfect reflected by e.g. current Wirkleistung of both electric smartmeter and inverted rectifier

bitstreamout, 2020-12-16:
Hmmm ... found this in ``htdocs/js/entity.js``

```
/**
 * Load total consumption from middleware
 * @return jQuery dereferred object
 */
Entity.prototype.loadTotalConsumption = function () {
        if (this.initialconsumption === undefined) {
                return $.Deferred().resolve().promise();
        }
        return vz.load({
                controller: 'data',
                url: this.middleware,
                identifier: this.uuid,
                context: this,
                data: {
                        from: 0,
                        tuples: 1,
                        group: 'day' // maximum sensible grouping level, first tuple dropped!
                }
        }).done(function (json) {
                // total observed consumption plus initial consumption value
                this.totalconsumption = (this.definition.scale || 1) * this.initialconsumption + json.data.consumption;
                // show in UI
                this.updateDOMRowTotal();
        });
};
```

if I read and understand this correct then the total consumption of each electric meter is always recalcuated from the previous calucated time and difference values tuples.   This is somehow strange ... not the calculated tuples but not to remember the latest value get from vzlogger

J-A-U, 2020-12-16:
Not so strange when you consider most channels in VZ don't actually have a consumtion to work with. It's a "one size fits all" solution.

bitstreamout, 2020-12-16:
Maybe an option not to calculate tuples for a graph for electric meter would be fine.  As owner of a PV I depend on (time) exact values for the filing of the VAT return ... nice graphs are secondary here (and shown by the electric effective power graph as well)

bitstreamout, 2020-12-16:
Meanwhile I see that the graph of Bezug has stopped at 8:30.  I've detected this by disabling the Wirkleistung which normally is exactly following the Bezug graph.
[image]

J-A-U, 2020-12-16:
> As owner of a PV I depend on (time) exact values for the filing of the VAT return

https://wiki.volkszaehler.org/software/frontends/meter_readings

> Meanwhile I see that the graph of Bezug has stopped at 8:30.

Logfile and config of vzlogger?

bitstreamout, 2020-12-16:
OK ...
[vzlogger.log](https://github.com/volkszaehler/volkszaehler.org/files/5702553/vzlogger.log)

[vzlogger.conf.txt](https://github.com/volkszaehler/volkszaehler.org/files/5702567/vzlogger.conf.txt)

bitstreamout, 2020-12-16:
Do you also need as well the bash scripts doing the curl and jq filter calls?

bitstreamout, 2020-12-16:
After stopping systemd service vzlogger, restarting both push-server as well as middleware service, and starting vzlogger again, the graph is back

J-A-U, 2020-12-16:
Nothing to pinpoint so far. Loglevel 5 isn't deep enough for a issue like that.

bitstreamout, 2020-12-16:
> Nothing to pinpoint so far. Loglevel 5 isn't deep enough for a issue like that.

Just see it again ... now I've restarted with loglevel 15

bitstreamout, 2020-12-17:
Had happen again, this morning with Wirkleistung as well as later on with the Bezug
[image]
[vzlogger.log.zip](https://github.com/volkszaehler/volkszaehler.org/files/5708123/vzlogger.log.zip)

bitstreamout, 2020-12-17:
> > As owner of a PV I depend on (time) exact values for the filing of the VAT return
> 
> https://wiki.volkszaehler.org/software/frontends/meter_readings
> 

Is there a currently working solution as this above is out of date (and I do not speak SQL)
[zaehlerwerte.php.txt](https://github.com/volkszaehler/volkszaehler.org/files/5708750/zaehlerwerte.php.txt)

I also can read port 8081 of vzlogger with the help of some jq code and curl

```
.data | .[] | select(.uuid == "65b7e2b0-3e1b-11eb-8ceb-3f89ed638d7f"),
              select(.uuid == "9a5e71e0-3e1b-11eb-9c1a-db732feba609"),
              select(.uuid == "b69eb5e0-3e1c-11eb-a050-b9764e30bec1") |
              "\(.uuid|sub("65b7e2b0-3e1b-11eb-8ceb-3f89ed638d7f";"Bezug")
                      |sub("9a5e71e0-3e1b-11eb-9c1a-db732feba609";"Einspeisung")
                      |sub("b69eb5e0-3e1c-11eb-a050-b9764e30bec1";"Erzeugung")),\(.tuples|.[])\n"
```

but alsoe here my experiences with jq is very humble like I miss knowledge on advanced looping or defining own functions

jahir, 2020-12-17:
I wrote a [perl script](https://gist.github.com/jahir/46539b7c8988b451c2ebeead24eb7da5) for the yearly reporting to my grid operator

J-A-U, 2020-12-18:
Minor:
`[Dec 17 08:34:52][chn6] CURL:   Trying ::1:8080...
[Dec 17 08:34:52][chn6] CURL: TCP_NODELAY set
[Dec 17 08:34:52][chn6] CURL: connect to ::1 port 8080 failed: Connection refused
[Dec 17 08:34:52][chn6] CURL:   Trying 127.0.0.1:8080...
[Dec 17 08:34:52][chn6] CURL: TCP_NODELAY set
[Dec 17 08:34:52][chn6] CURL: Connected to localhost (127.0.0.1) port 8080 (#1777)
`
Destination localhost for your exec-channels doesn't work properly.

J-A-U, 2020-12-18:
There is something very strange going on with your timestamps. Sending the sml-channels to middleware always works. But the time in tuples differs somehow from the time in log.
E.g. good:
`[Dec 17 03:59:04][chn0] CURL: Sent '[ [ 1608173828682, 439996.20000000001 ], ... , [ 1608173944751, 440000.80000000005 ] ]' bytes`
1608173944751 = Donnerstag, 17. Dezember 2020 03:59:04.751 GMT+01:00

Bad:
`[Dec 17 09:55:05][chn0] CURL: Sent '[ [ 1608194059441, 441839.70000000001 ], ... , [ 1608194183639, 441853.5 ] ]' bytes`
1608194183639 = Donnerstag, 17. Dezember 2020 09:36:23.639 GMT+01:00

But your last reading from meter was fine:
`[Dec 17 09:55:05][chn0] Adding reading to queue (value=441962.90 ts=1608195305342)`
1608195305342 = Donnerstag, 17. Dezember 2020 09:55:05.342 GMT+01:00

My best guess:
Your unusual configuration (interval usually used for pull) makes vzlogger go crazy with the timestamps in queue. Should look more like this.
```
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "read_timeout": 10,
      "aggtime": 10,
      "protocol": "sml",
      "device": "/dev/ttyUSB0",
      "channels": [{
          "uuid": "65b7e2b0-3e1b-11eb-8ceb-3f89ed638d7f",
          "middleware": "http://127.0.0.1:8080",
          "identifier": "1-0:1.8.1",
        "aggmode": "max",
          "scaler": 1,
        },{
          "uuid": "9a5e71e0-3e1b-11eb-9c1a-db732feba609",
          "middleware": "http://127.0.0.1:8080",
          "identifier": "1-0:2.8.1",
        "aggmode": "max",
          "scaler": 1,
        },{
          "uuid": "c947d780-3e1b-11eb-bba8-5d89638d37c5",
          "middleware": "http://127.0.0.1:8080",
          "identifier": "1-0:16.7.0",
          "scaler": 1,
        }],
    },
```

bitstreamout, 2020-12-18:
This missing or ruptured  graphs seems to be an issue with the graphical layer as if I switch back one day the graphs are there again.  IMHO this is not a problem of the vzlogger but in the handling of the tuples data them selfs but during plotting those tuples the algorithm seems to get lost under some  circumstances. Ive no idea *what* circumstances this are but it is noticeable that only data from the *sml* channels are affected ... and very likly those where the steps of the values between successive tuples are very high.

frankrichter, 2020-12-18:
Please try again with stopped push server. Push updates prevent the
frontend from auto-updating.

bitstreamout, 2020-12-19:
> I wrote a [perl script](https://gist.github.com/jahir/46539b7c8988b451c2ebeead24eb7da5) for the yearly reporting to my grid operator

Thanks a lot!  I've done some small changes ... beside name of the db and password ... I have done

```
-    my $per0 = (shift @$ints)->[0];
+    my $per0 = (shift @$ints)[0];
     my $tb = Text::Table->new(qw(year reading kWh));
     foreach my $int (@$ints) {
+print STDERR "Im here\n";
         my $per = $int->[0];
         my $tsms = $int->[1];
         my ($val) = $dbh->selectrow_array(q(select value from data where channel_id=? and timestamp=?), undef, $chid, $tsms);
@@ -36,5 +38,6 @@ if (not defined $chid) {
         $per0 = $per;
         $val0 = $val;
     }
+    $dbh->disconnect();
     print $tb;
 }
```

this samll tool list all meter uuids but specifying a uuid results in

 ```
smart@smartmeter:~> bin/meter.pl 65b7e2b0-3e1b-11eb-8ceb-3f89ed638d7f
year reading kWh
```
Is there a way to specify not only the uuid but also a date like today with 2020-12-19 and get back the last raw tuple of the day?
With this it would be very simple to modify the attached php script (based on Rudis php script) in such a way to get VAT relevant data to do the mounthly VAT pre report via elster to the tax office
[zaehlerwerte.php.txt](https://github.com/volkszaehler/volkszaehler.org/files/5719090/zaehlerwerte.php.txt)

this php is used with thttpd hence the wrapper script zaehlerwerte.cgi

bitstreamout, 2020-12-19:
Just an other observation  ... the php function register_shutdown_function() seems not to work well for stopping the middleware systemd service unit as it left over many ``/tmp/dbg<RANDOM>``  files after restart or stop:

```
smart@smartmeter:~/volkszaehler> grep -B2 -rsE 'register_shutdown_function.*unlink'  .
Binary file ./composer.phar matches
--
./vendor/symfony/process/PhpProcess.php-            $file = tempnam(sys_get_temp_dir(), 'dbg');
./vendor/symfony/process/PhpProcess.php-            file_put_contents($file, $script);
./vendor/symfony/process/PhpProcess.php:            register_shutdown_function('unlink', $file);
--
./vendor/php-pm/php-pm/src/ProcessManager.php-        $file = tempnam(sys_get_temp_dir(), 'dbg');
./vendor/php-pm/php-pm/src/ProcessManager.php-        file_put_contents($file, $script);
./vendor/php-pm/php-pm/src/ProcessManager.php:        register_shutdown_function('unlink', $file);
```

frankrichter, 2020-12-19:
Try middleware requests with &options=raw to get raw meter readings. No
need to mess around with the database.

bitstreamout, 2020-12-19:
Hmm ... has taken a while but ``'http://smartmeter.intern.lan:8080/data.json?options=raw&uuid%5B%5D=<uuid>...`` delivers a tuple which is definitively not the last submitted/polled value of Bezug, it is much to less:

```
      "tuples": [
        [
          1608291569817,
          451316.1,
          1
        ]
      ],
```

but asking the vzlogger on port 8081 shows:

```
      "tuples": [
        [
          1608378080341,
          460318.9
        ]
      ]
```

bitstreamout, 2020-12-19:
Playing around and found that

```
bc -l <<< $(date +%s)*1000
```

and use this value for ``from=<ts>`` does give back the last values :+1:

frankrichter, 2020-12-19:
You can use &from=now

bitstreamout, 2020-12-19:
> You can use &from=now

Meanwhile I've detected this ... and moved my small php script forward to use exec() with a helper script doing curl+jq on middleware data.json.  Seems to work
[image]

bitstreamout, 2020-12-19:
Found that php extension curl is very helpful but the extension for json with its json_decode() is useless for handling json data get from middleware server.

andig, 2020-12-19:
Welches Problem soll hier eigentlich gelöst werden?

bitstreamout, 2020-12-20:
Korrekte Anzeige der Zählerwerte .... was mit dem Algorithmus in Entity.prototype.loadTotalConsumption nicht gegeben ist.  Das kleine php Skript ist mein Versuch korrekte, d.h. umsatzsteuerrelevante Zählerwerte zu bekommen. D.h. Strombezug, Stromeinspeisung, Stromerzeugung und die daraus resultierende Werte wie Eigenverbrauch[1] und Gesamtverbrauch zu bestimmen. 

[1] Der Eigenverbrauch ist steuerrechtlich eine Entnahme, welche der PV-Anlagenbetreiber als Unternehmer für den Eigenbedarf vornimmt und daher muss für die Entnahme die gleiche Umsatzsteuer wie für den privaten Strombezug an das Finanzamt monatlich abgeführt werden.  Daneben muss natürlich auch die Umsatzsteuer für die Einspeisevergütung abegührt werden.

J-A-U, 2020-12-20:
Well, by adding another problem to the thread i totally missed this.
> Is there a currently working solution as this above is out of date (and I do not speak SQL)

The working solution is linked in the article, and Frank pointed to the needed details. You get all the 1:1 readings out of middleware.

> but the extension for json with its json_decode() is useless for handling json data

Example:
```
      $content = curl_file_get_contents($url);
      $content=json_decode($content);
      if (!empty($content->data->tuples)) {
        $lastTuple = end($content->data->tuples);
        $lastValue = $lastTuple[1];
      } // if
```

andig, 2020-12-20:
> Korrekte Anzeige der Zählerwerte .... was mit dem Algorithmus in Entity.prototype.loadTotalConsumption nicht gegeben ist. 

Ich denke hier geht es ein wenig durcheinander. Erstmal dürften für die steuerrelevanz korrekte Daten notwendig sind. Das sind vmtl. Zählerstände. Die stehen dem Frontend hier aber in keiner Weise zur Verfügung (können aber aus der Datenbank ausgelesen werden).

Davon abgesehen: was gibt es konkret am Algorithmus zu bemängeln?

Die ganzen PHP Beispiele haben m.E. in diesem Ticket nichts zu suchen- für Diskussionen bitte die Mailingliste verwenden.

bitstreamout, 2020-12-20:
> Erstmal dürften für die steuerrelevanz korrekte Daten notwendig sind. Das sind vmtl. Zählerstände. Die stehen dem Frontend hier aber in keiner Weise zur Verfügung (können aber ~aus der Datenbank~ per Middleware API ausgelesen werden). Damit sollten alle notwendigen Informationen für eine Abrechnung vorliegen.
> 
> Davon abgesehen: was gibt es konkret am Algorithmus zu bemängeln?

Die von der middleware **aktuell** angezeigten Werte für die Zähler sind kleiner als die **letzen** Werten des vzloggers für die selben Zähler.  Und das nicht nur ein klein wenig, sondern teilweise um mehrere kWh, also keine Rundungs oder Skalierungsfehler.

bitstreamout, 2020-12-20:
Beispiel von gerade eben ``Sun Dec 20 13:20:56 CET 2020`` ... das sind keine Rundungsfehler:

[image]

frankrichter, 2020-12-20:
Die Zählerstände im Frontend werden gerechnet ab dem ersten Tageswechsel,
was davor war fällt unter den Tisch. Initialwert einmalig entsprechend
anpassen, dann passt das zukünftig.

Magst du deine Lösung vielleicht ins Wiki stellen?

bitstreamout, 2020-12-25:
> Die Zählerstände im Frontend werden gerechnet ab dem ersten Tageswechsel, was davor war fällt unter den Tisch. Initialwert 
> einmalig entsprechend anpassen, dann passt das zukünftig. Magst du deine Lösung vielleicht ins Wiki stellen?

Ich habe das php-Skript als auch die Helferlein in [zaehlerwerte](https://github.com/bitstreamout/zaehlerwerte) abgelegt

closed 2020-12-25 by andig