volkszaehler/volkszaehler.org #839: Network error - Not found

torsten3009, 2021-01-18:
ich komme leider nicht wirklich weiter...

Zählertyp: Q3m easymeter
Plattform : raspberry buster lite
vzlogger Version vzlogger 0.8.0 - volkszaehler.org logging utility
vzlogger.log liefert korrekte daten
Middleware/ Frontend Version (git log -n 1 im Ordner /var/www/volkszaehler.org)
`    pi@iobroker:/var/www/volkszaehler.org $ git log -n 1
    commit 61bf9f5a5b1e434d50781c2c620bd531c2576efa (HEAD -> master, origin/master, origin/HEAD)
    Merge: 7ff47a1 54849fb
    Author: andig <cpuidle@gmx.de>
    Date:   Sat Nov 14 10:45:10 2020 +0100

        Merge pull request #788 from SirkoVZ/master

        Added absolute humidity`

auf der Seite steht nur Network error - Not found
Beim anlegen eines Kanal: Uncaught TypeError: Cannot read property 'forEach' of undefined

Alles bereits mehrfach neu installiert. Datenbank und user sollten passen.

Welche logs sollte ich mir noch ansehen?

J-A-U, 2021-01-18:
Für normalen Service wäre die Mailingliste besser, wir versuchen im git nur echte Codefragen zu behandeln.

Hast du als Basis das Image genommen oder selbst installiert? Was für einen Webserver nutzt du? PPM?

mfg JAU

torsten3009, 2021-01-18:
Danke mit apache 2 läuft es jetzt...

Echt nicht soooo einfach umzusetzen

Danke!

Viele Grüße
Torsten 

Viele Grüße, Torsten

J-A-U, 2021-01-18:
Das ist vor allem die Herausforderung der korrekten Anleitung zu folgen, auf der Wikiseite werden ja mehrere Varianten vorgeschlagen.

Und schließe bitte noch das Issue.

closed 2021-01-19 by andig