volkszaehler/volkszaehler.org #848: network error demo.volkszaehler.org when accessing volkszaehler gui

oriziena, 2021-03-21:
Hi @all,

hopefully this is not a completely dumb issue...

afterupgrading to latest volkszaehler release I am experiencing the issue, that the demo page stuff seems not to be reachable anymore. is there a way to deactivate that ?
[image]

How to fix that stuff? thanks
BR
Oriziena

J-A-U, 2021-03-21:
https://github.com/volkszaehler/volkszaehler.org/blob/c0634e45fddedd8baa4a0c0a2c5248a09cd1489a/htdocs/js/options.js#L49

https://www.photovoltaikforum.com/thread/153815-network-error-error/

mfg JAU

oriziena, 2021-03-21:
hi,

okay... that was fast, easy and efficient...

therefore closing it ..

thanks @J-A-U

closed 2021-03-21 by oriziena