volkszaehler/volkszaehler.org #850: Docker Build fails due to wrong php Version

itchy2, 2021-03-26:
Hi all,

I tried to use the dockerbuild from the GIT repository but in the build environment seems to be an issue with the PHP Version.

Output attached:

Sending build context to Docker daemon   13.2MB

Step 1/15 : FROM composer:1 AS builder
 ---> 1320a67add67
Step 2/15 : WORKDIR /vz
 ---> Using cache
 ---> e9e0c7a6a118
Step 3/15 : COPY composer.json /vz
 ---> Using cache
 ---> 5f13dbec88ca
Step 4/15 : RUN composer install --no-ansi --no-scripts --no-dev --no-interaction --no-progress --optimize-autoloader
 ---> Running in 9c0f4fc948d3
[91mLoading composer repositories with package information
[0m[91mUpdating dependencies
[0m[91mYour requirements could not be resolved to an installable set of packages.

  Problem 1
    - php-pm/php-pm 2.2.1 requires ext-pcntl * -> the requested PHP extension pcntl is missing from your system.
    - php-pm/php-pm 2.2.0 requires ext-pcntl * -> the requested PHP extension pcntl is missing from your system.
    - php-pm/php-pm 2.1.3 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.1.2 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.1.1 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.1.0 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.0.4 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.0.3 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.0.2 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.0.1 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - php-pm/php-pm 2.0 requires php ^7.1 -> your PHP version (8.0.3) does not satisfy that requirement.
    - Installation request for php-pm/php-pm ^2.0 -> satisfiable by php-pm/php-pm[2.0, 2.0.1, 2.0.2, 2.0.3, 2.0.4, 2.1.0, 2.1.1, 2.1.2, 2.1.3, 2.2.0, 2.2.1].

  To enable extensions, verify that they are enabled in your .ini files:
    - /usr/local/etc/php/php-cli.ini
    - /usr/local/etc/php/conf.d/docker-php-ext-sodium.ini
    - /usr/local/etc/php/conf.d/docker-php-ext-zip.ini
  You can also run `php --ini` inside terminal to see which files are used by PHP in CLI mode.
Running update with --no-dev does not mean require-dev is ignored, it just means the packages will not be installed. If dev requirements are blocking the update you have to resolve those problems.
[0m

r00t-, 2021-03-27:
the dockerfile uses `composer:1`
which moved from php7 to php8 on jan 30:
https://github.com/docker-library/repo-info/commit/8bdf2db8a738920d66ce5b0a0fbe8816232ebde7#diff-1086562996d7bc7b5c2320bab4070380f26f47d35535fa517efbe05471b5905dL20

i don't care about composer or it's docker image, but having a versioned tag not refer to a fixed versions seems bad design.

go find an image that still has php7

J-A-U, 2021-03-27:
Not long ago Andi added support for PHP8: https://github.com/volkszaehler/volkszaehler.org/pull/840

Therefore it seems more suitable to fix ext-pcntl. Because php-pm ^2.2.0 also supports PHP8.

itchy2, 2021-03-27:
> go find an image that still has php7

The images for the build process are part of the GIT repository: https://github.com/volkszaehler/volkszaehler.org/blob/master/Dockerfile 
I just did a checkout and started the docker build. To me it looks like a wrong version in one of the used images for the docker build.

KR
Itchy2

J-A-U, 2021-03-28:
I have a question, because i'm not familiar with docker.
Is `- /usr/local/etc/php/conf.d/docker-php-ext-sodium.ini` a local file on your machine?

itchy2, 2021-03-28:
Hi J-A-U

the error occurs in Step 4. Step 4 can be seen in the Docker File Line 9 here: https://github.com/volkszaehler/volkszaehler.org/blob/c0634e45fddedd8baa4a0c0a2c5248a09cd1489a/Dockerfile#L9

The file sodium.ini is part of the composer docker image here: https://hub.docker.com/_/composer?tab=tags&page=1&ordering=last_updated with the tag 1.

Does that answer your question?

Kind regards
Itchy2

J-A-U, 2021-03-29:
It sort of answers it.

My best guess: We need something like Line 21 before Line 9.

closed 2021-03-31 by andig