volkszaehler/volkszaehler.org #860: Change port of middleware from 80 to 52004, VZlogger throws context issue

NetDevils, 2021-06-04:
Hi there,
love working with the Volkszaehler !! Great piece of software !! Helped me and a neighbor to monitor the power consumption of our heating system and by that troubleshooting installation failures. Now we'll be able to save several hundred Euros of energy costs per year! Perfect!

Now I would like to make it a permanent solution and move the connection away from port 80 to 52004 to make room for other services. But the Apache config and VZLogger cURL commands keep failing :-/

- middleware.json contains "port": 52004 👍 
- WebUI is available at http://hostname:80 as well as http://hostname:52004 👍 
- API POSTs at http://hostname:52004/77e671f0-c3c9-11eb-a5c7-8b2887386909.json?ts=1622656942000&value=888.88 are working 👍 

My ~/volkszaehler.org/htdocs/.htaccess contains:
```
<IfModule mod_proxy.c>
        RewriteEngine On
        RewriteRule ^middleware(.php)?(/.*)? http://localhost:52004/$2 [P]
        RewriteRule ^api(/.*)$ http://localhost:52004/$1 [P]
        RewriteRule (.*) http://localhost:52004/$1 [P]
</IfModule>
```

If my /etc/vzlogger.conf contains:
```
"meters" : [
        {
                "enabled" : true,
                (...)
                "channels": [{
                   "uuid" : "44eb2190-86a7-11eb-8771-83a104f486d3",
                   "middleware" : "http://localhost/",
                   "(...)
                }]
        }
]
```
everything is working fine 👍 

**_However_**, if my /etc/vzlogger.conf contains:
```
"meters" : [
        {
                "enabled" : true,
                (...)
                "channels": [{
                   "uuid" : "44eb2190-86a7-11eb-8771-83a104f486d3",
                   "middleware" : "http://localhost:52004/",
                   "(...)
                }]
        }
]
```
the data is not passed through 👎
and the log says:

> [Jun 03 19:28:59][chn0] CURL:   Trying 127.0.0.1:52004...
> [Jun 03 19:28:59][chn0] CURL: TCP_NODELAY set
> [Jun 03 19:28:59][chn0] CURL: Connected to localhost (127.0.0.1) port 52004 (#0)
> [Jun 03 19:28:59][chn0] CURL: Sent 41 bytes: '[ [ 1622741330000, 14692203.422229633 ] ]'
> [Jun 03 19:28:59][chn0] CURL: upload completely sent off: 41 out of 41 bytes
> [Jun 03 19:28:59][chn0] CURL: Mark bundle as not supporting multiuse
> [Jun 03 19:28:59][chn0] CURL: Received 87 bytes: '{"version":"0.3","exception":{"message":"Missing context","type":"Exception","code":0}}'
> [Jun 03 19:28:59][chn0] CURL: Closing connection 0
> [Jun 03 19:28:59][chn0] CURL Error from middleware: 'Exception': 'Missing context'

I feeld like I have tried everything by now but I just don't get why VZlogger is not able to send the data to the middleware after I added the port ?!? The API and WebUI are fine with the port....

Is it with the RewriteRules?
In the end I will also want to switch to https (I already have other services running on https, thus no worries about the certs) and will need to deactivate the port 80 instance all together. But only once all and every traffic is running on port 52004 of course.

Thanks for reading up until here and any hint is highly welcome.

Cheers

r00t-, 2021-06-05:
this is for reporting bugs in the software, it is not a user support forum.

the error message indicates that no UUID could be parsed from the request:
https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Router.php#L138

you should include the full request as sent by vzlogger (from it's log) and as received by the webserver (from it's logfile),
possibly including the mod_rewrite debug log.
also, what were the rewrite rules you were using for the working port 80 configuration?

J-A-U, 2021-06-05:
> and move the connection away from port 80 to 52004

Well, what you actually did was moving the connection from Port 8080 to 52004.

In VZ-image on port 8080 the middleware (running on php-pm) is answering. If you actually want to move away from port 80 you have to change Apache config.

closed 2021-06-05 by andig

NetDevils, 2021-06-05:
@r00t- : would have helped to know were to ask for support then...
@J-A-U : thanks for the hint. I did change to middleware to 42004 now, changed Apache config to listen on 52004 and changed the RewriteRules in htaccess to forward to :42004. Now I can access the UI on 52004, have the middleware run on 42004 and have both port 80 and 8080 free for other websevers to run. thanks!