volkszaehler/volkszaehler.org #866: composer installiert fehlerahftes php-pm 2.2.1

StefanSchoof, 2021-07-15:
php-pm 2.2.1 hat ein Problem, welches in https://github.com/php-pm/php-pm/issues/538 beschrieben wird. Es gibt als fix Version [2.2.2](https://github.com/php-pm/php-pm/releases/tag/2.2.2)

Wenn ich in docker ein build mache wird merkwürdigerweise Version 2.2.1 genommen:
```
#14 0.932 Loading composer repositories with package information
#14 2.252 Updating dependencies
#14 2.453 Lock file operations: 111 installs, 0 updates, 0 removals
...
#14 2.455   - Locking php-pm/httpkernel-adapter (2.0.6)
#14 2.455   - Locking php-pm/php-pm (2.2.1)
...
```

Dies führt zu genau diesem Fehler und 500 internal Server Error Antworten.

In https://github.com/volkszaehler/volkszaehler.org/blob/master/composer.json#L26 wird `^2.0` angegeben. Ich verstehe nicht warum composer nicht [2.2.2](https://packagist.org/packages/php-pm/php-pm#2.2.2) nimmt.

Ein Fix bei mir war jetzt:
```
        "php-pm/php-pm": "^2.2.2",
```

Da ich das ganze nicht vollständig verstehe, bin ich mir nicht sicher, ob das als PR machen soll oder ob ich gerade einen doofen Fehler mache.

andig, 2021-07-16:
Verstehe ich leider auch nicht. Es müsste die letzte Version sein.

StefanSchoof, 2021-07-16:
Ich denke ich verstehe jetzt was passiert:
Es gibt auch noch
https://github.com/volkszaehler/volkszaehler.org/blob/b2e4cd9053275cc38d8400a2c76b08e96df486f1/composer.json#L24

composer hat also die Aufgabe die höchste mögliche Version von php-pm und react/http zu installieren. Und er entschiedet sich für react/http in 1.4 und dann php-pm nur in 2.2.1 zu nehmen. Mir fallen jetzt 3 Möglichkeiten für eine Lösung ein:
1. react/http als abhängigkeit aus volkszaheler zu löschen. Es wird als Abhängigkeit von php-pm in Version 1.2.0 installier
2. Die gleiche Zeile wie php-pm nutzen: `"react/http": ">=1.0 <1.3"` https://github.com/php-pm/php-pm/blob/63430a72843b6cbf73383f32d3dd614db18bb4d1/composer.json#L12
3. `"php-pm/php-pm": "^2.2.2"`

Ich halte gerade 1. für die beste Lösung. Dann sollte, wenn php-pm das Problem löst automatisch die höhere Version von react/http installiert werden. (Wenn wir das heute noch entschieden mache ich noch schnell ein PR. Sonst bin ich erstmal eine Woche ohne PC im Urlaub)

andig, 2021-07-16:
Macht keinen Sinn. Warum sollte ppm reaact/http 1.4 verhindern? zur Sicherheit: Du startest ohne oder mit sauberem composer.lock file?

andig, 2021-07-16:
Entschuldige. Alles nochmal gelesen- dann lass uns react/http aus vz entfernen.

PR #867 (2021-07-16): Remove react/http dependency
react/http is a depency of php-pm/php-pm and stil will be installed
react/http 1.4 and php-pm/php-pm 2.2.2 have problems, see https://github.com/php-pm/php-pm/issues/538
Without this react/http 1.4 and php-pm 2.2.1 will be insatlled and result into that issue

Resolves #866
commit 601f3e83 (2021-07-16): Remove react/http dependency (#867)

react/http is a depency of php-pm/php-pm and stil will be installed
react/http 1.4 and php-pm/php-pm 2.2.2 have problems, see https://github.com/php-pm/php-pm/issues/538
Without this react/http 1.4 and php-pm 2.2.1 will be insatlled and result into that issue

Resolves #866
closed 2021-07-16 by andig