volkszaehler/volkszaehler.org #870: Cache update results in non working middleware

dm82m, 2021-08-08:
Hey guys,

with the commit 4d0196a153aa8750d72c9d7906e5f4bf92ed13a5 my Volkszaehler stops working:

```
pi@raspberrypi:~/volkszaehler.org $ sudo composer update
Do not run Composer as root/super user! See https://getcomposer.org/root for details
Loading composer repositories with package information
Warning from https://repo.packagist.org: Support for Composer 1 is deprecated and some packages will not be available. You should upgrade to Composer 2. See https://blog.packagist.com/deprecating-composer-1-support/
Updating dependencies (including require-dev)
Package operations: 0 installs, 1 update, 0 removals
  - Updating doctrine/cache (1.12.1 => 2.1.1): Loading from cache
Writing lock file
Generating autoload files
composer/package-versions-deprecated: Generating version class...
composer/package-versions-deprecated: ...done generating version class
pi@raspberrypi:~/volkszaehler.org $ sudo systemctl restart middleware
```

Within UI it shows this error afterwards:
[image]

Any idea what I can do?

Best,
Dirk

andig, 2021-08-08:
Gibts Einträge im Apache errorlog? Falls der Commit schuld wäre müsste es ja auf der PHP-Seite schief gehen. Gehts wieder wenn Du einen Commit zurück gehst?

dm82m, 2021-08-08:
Ja, geht wieder wenn ich den vorletzten Commit nehme - reproduzierbar. Ich nute das fertige Raspberry Image, da scheint kein Apache2 zu laufen. Wo könnte ich sonst noch relevante Logs finden?

andig, 2021-08-08:
Syslog? Ich rate aber…

dm82m, 2021-08-09:
php loggt in syslog ja, aber nach doctrine/cache update und zugriff im browser mit der obigen fehlermeldung wird nichts sinnvolles geloggt:

```
Aug  9 08:42:54 raspberrypi systemd[1]: Started Session 504 of user pi.
Aug  9 08:43:26 raspberrypi php[18946]: Server is shutting down.
Aug  9 08:43:26 raspberrypi systemd[1]: Stopping Volkszaehler.org Middleware...
Aug  9 08:43:26 raspberrypi php[18946]: Stopping the process manager.
Aug  9 08:43:26 raspberrypi systemd[1]: middleware.service: Succeeded.
Aug  9 08:43:26 raspberrypi systemd[1]: Stopped Volkszaehler.org Middleware.
Aug  9 08:43:26 raspberrypi systemd[1]: Started Volkszaehler.org Middleware.
Aug  9 08:43:26 raspberrypi php[29689]: Read configuration /home/pi/volkszaehler.org/etc/middleware.json, modified by command arguments.
Aug  9 08:43:26 raspberrypi php[29689]: /
Aug  9 08:43:26 raspberrypi php[29689]: +---------------------------+-----------------------------------------+
Aug  9 08:43:26 raspberrypi php[29689]: | host                      | 0.0.0.0                                 |
Aug  9 08:43:26 raspberrypi php[29689]: | port                      | 8080                                    |
Aug  9 08:43:26 raspberrypi php[29689]: | workers                   | 8                                       |
Aug  9 08:43:26 raspberrypi php[29689]: | static-directory          | /var/www/volkszaehler.org/htdocs        |
Aug  9 08:43:26 raspberrypi php[29689]: | logging                   | 0                                       |
Aug  9 08:43:26 raspberrypi php[29689]: | max-requests              | 1000                                    |
Aug  9 08:43:26 raspberrypi php[29689]: | bridge                    | HttpKernel                              |
Aug  9 08:43:26 raspberrypi php[29689]: | bootstrap                 | Volkszaehler\Server\PPMBootstrapAdapter |
Aug  9 08:43:26 raspberrypi php[29689]: | app-env                   | dev                                     |
Aug  9 08:43:26 raspberrypi php[29689]: | debug                     | 0                                       |
Aug  9 08:43:26 raspberrypi php[29689]: | max-execution-time        | 30                                      |
Aug  9 08:43:26 raspberrypi php[29689]: | memory-limit              | -1                                      |
Aug  9 08:43:26 raspberrypi php[29689]: | limit-concurrent-requests |                                         |
Aug  9 08:43:26 raspberrypi php[29689]: | request-body-buffer       |                                         |
Aug  9 08:43:26 raspberrypi php[29689]: | ttl                       | 0                                       |
Aug  9 08:43:26 raspberrypi php[29689]: | populate-server-var       | 1                                       |
Aug  9 08:43:26 raspberrypi php[29689]: | socket-path               | .ppm/run/                               |
Aug  9 08:43:26 raspberrypi php[29689]: | pidfile                   | .ppm/ppm.pid                            |
Aug  9 08:43:26 raspberrypi php[29689]: | reload-timeout            | 30                                      |
Aug  9 08:43:26 raspberrypi php[29689]: | cgi-path                  | /usr/bin/php                            |
Aug  9 08:43:26 raspberrypi php[29689]: +---------------------------+-----------------------------------------+
Aug  9 08:43:26 raspberrypi php[29689]: Starting PHP-PM with 8 workers, using StreamSelectLoop ...
Aug  9 08:43:27 raspberrypi php[29689]: 8 workers (starting at 5501) up and ready. Application is ready at http://0.0.0.0:8080/
```

andig, 2021-08-09:
Probier mal bitte den Request im Browser direkt zu machen. Also in der KOnsole schauen was da aufgerufen wird und dann mal in die Addresszeile eingeben. Wenn das noch nciht reicht auch ein &debug=1 anhängen. Irgendwo muss eine Fehlermeldung sichtbar sein.

dm82m, 2021-08-09:
Folgendes erhalte ich im Browser beim direkten Aufruf des Calls:

`{"version":"0.3","exception":{"message":"Class 'Doctrine\\Common\\Cache\\ArrayCache' not found","type":"Error","code":0}}`

andig, 2021-08-09:
Schau mal bitte ob das tut

dm82m, 2021-08-09:
```
PHP Fatal error:  Uncaught Error: Class 'Symfony\Component\Cache\Adapter\ArrayAdapter' not found in /home/pi/volkszaehler.org/lib/Router.php:238
Stack trace:
#0 /home/pi/volkszaehler.org/bin/aggregate(71): Volkszaehler\Router::createEntityManager(true)
#1 /home/pi/volkszaehler.org/bin/aggregate(285): BasicCommand->__construct()
#2 {main}
  thrown in /home/pi/volkszaehler.org/lib/Router.php on line 238
```

andig, 2021-08-09:
Dann probier mal bitte symfony/cache 5.3?

dm82m, 2021-08-09:
mit 5.3 gibts keinen php error mehr aber dafür wieder:

`{"version":"0.3","exception":{"message":"Class 'Doctrine\\Common\\Cache\\ArrayCache' not found","type":"Error","code":0}}`

jahir, 2021-08-09:
Sehe das gleich (...ArrayCache not found) mit 5.3. Was bei mir dann aber geholfen hat (von [SO](https://stackoverflow.com/a/68213732/6735159)):

`composer require doctrine/cache:^1.11`

Mittelfristig sollten wir aber wohl schon doctrine/cache ablösen, wenn das deprecated ist...

andig, 2021-08-09:
…also entweder revert Commit oder Lösung finden

dm82m, 2021-08-09:
ich springe mal zurück auf master - 1 und wenn ich noch mal was testen soll sag einfach bescheid

jahir, 2021-08-09:
Das ist ja ein übles Version-Wirrwarr. Das doctrine-Projekt selbst stuft doctrine/cache als deprecated ein und sagt, man solle was anderes benutzen. Gleichzeitig erfordern dbal, orm und persistance cache. Das läßt sich bei uns wohl nicht lösen, oder?

jahir, 2021-08-09:
> Das läßt sich bei uns wohl nicht lösen, oder?

Oh, wahrscheinlich doch... Soweit ich das verstanden habe, enthält doctrine/cache ab 2.0 nur noch die interfaces (für backwards compatibility), aber keine Implementation mehr. Dafür gibt es dann z.B. symfony/cache, dafür muß man aber seinen code etwas umstellen. Ist in einem [PR](https://github.com/doctrine/cache/pull/391/files) bzw. [hier](https://github.com/beberlei/cache/blob/DocsAlternatives/docs/en/index.rst) und [hier](https://github.com/beberlei/cache/blob/DocsAlternatives/UPGRADE-1.11.md) beschrieben.

andig, 2021-08-09:
Das hab ich im PR ja versucht- leider ohne Erfolg. Auf jeden Fall hat Doctrine da klare Kommunikation vermissen lassen...

jahir, 2021-08-09:
Hm, damit tut's bei mir mit doctrine/cache 2.x: https://github.com/volkszaehler/volkszaehler.org/pull/872

jahir, 2021-08-09:
@dm82m Könntest du nochmal #871 testen?

closed 2021-08-10 by andig