volkszaehler/volkszaehler.org #887: capabilities/database.json? wirft Exception

StefanSchoof, 2021-12-20:
Beim Aufruf von `capabilities/database.json?` kommt eine Fehlermeldung. Tritt auch auf wenn man dies bei dem demo Server macht: [https://demo.volkszaehler.org/middleware.php/capabilities/database.json?](https://demo.volkszaehler.org/middleware.php/capabilities/database.json?)

```json
{"version":"0.3","exception":{"message":"An exception occurred while executing 'EXPLAIN SELECT COUNT(id) FROM data USE INDEX (PRIMARY)':\n\nSQLSTATE[42S22]: Column not found: 1054 Unknown column 'id' in 'field list'","type":"InvalidFieldNameException","code":0}}
```

Denke es hat die gleich Ursache wie https://github.com/volkszaehler/volkszaehler.org/issues/886#issuecomment-990737145

Das Wiki erwähnt die URL unter https://wiki.volkszaehler.org/howto/datenmengen#aggregation_pruefen

StefanSchoof, 2021-12-20:
Die Zeile ist vermutlich 
https://github.com/volkszaehler/volkszaehler.org/blob/7225eac0a6582b4b8ae25a60d5cc506310fabddb/lib/Controller/CapabilitiesController.php#L47

andig, 2021-12-20:
Yep, das müsste jetzt `count(*)` sein, PR welcome

StefanSchoof, 2021-12-21:
Ja, mit `COUNT(*)` geht die Seite wieder.

PR #888 (2021-12-21): id column no longer extsis => use * for count
resolves #887
closed 2021-12-21 by andig