volkszaehler/volkszaehler.org #891: Missing doctrine/annotations dependency

StefanSchoof, 2022-01-15:
Wenn ich ohne composer.lock datei eine volkzaehler starte bekomme ich seit kurzen die Fehlermeldung:

```
Fatal error: Uncaught Error: Class "Doctrine\Common\Annotations\AnnotationRegistry" not found in /vz/vendor/doctrine/orm/lib/Doctrine/ORM/Configuration.php:165
Stack trace:
#0 /vz/lib/Router.php(242): Doctrine\ORM\Configuration->newDefaultAnnotationDriver('/vz/lib/Model')
#1 /vz/bin/doctrine(31): Volkszaehler\Router::createEntityManager(true)
#2 {main}
thrown in /vz/vendor/doctrine/orm/lib/Doctrine/ORM/Configuration.php on line 165
```

Ich konnte das lösen indem ich wie unter https://stackoverflow.com/a/56858362/4319676 doctrine/annotations Abhängigkeit einbaue.

```json 
    "require": {
        ...
        "doctrine/annotations": "^1.13.2",
        ...
     }
```

Ich verstehe nicht warum das jetzt zu einem Fehler führt und vorher ging, vermute aber das Updates von Abhängigkeiten dazu geführt haben.

Sollen wir die Abhängigkeit hinzufügen oder da es ähnlich wie #866 wahrscheinlich durch minor updates der Abhängigkeiten entstanden ist, eine composer.lock mit einzuchecken um sicher zu gehen, dass Versionen installiert werden die zusammen arbeiten?

andig, 2022-01-16:
Warum solltest du ohne composer.lock starten? D.h. du hast nix installiert…

StefanSchoof, 2022-01-16:
Ja, frisch aus git ausgecheckt.

StefanSchoof, 2022-01-16:
Die composer Docs sagen: 
> Committing this file to version control is important  ...
https://getcomposer.org/doc/01-basic-usage.md#commit-your-composer-lock-file-to-version-control
Aber hier ist die compser.lock ist in der gitignore https://github.com/volkszaehler/volkszaehler.org/blob/650f5dfed07a370b4dacb479f8eea0137e648cbf/.gitignore#L4

andig, 2022-01-16:
Es ist egal, ob Du die Dependencies mit oder ohne composer.lock nicht installiert hast. Bei Dir fehlt einfach ein wesentlicher Installationsschritt…

closed 2022-01-16 by andig

StefanSchoof, 2022-01-16:
Ich denke du hast das Problem nicht verstanden. 
Für mich habe ich das Problem gelöst. Ich benötige keine Hilfe, sondern möchte nur das andere nicht auch noch in das Problem laufen.

Wenn man zur aktuellen Zeit eine composer.lock neu erstellt, fehlt darin die `doctrine/annotations` und Volkszaehler läuft so nicht.

andig, 2022-01-16:
> Ich denke du hast das Problem nicht verstanden.

Wenn Du es wie eben beschrieben hättest hätte das vllt geklappt. Aus

> Wenn ich ohne composer.lock datei eine volkzaehler starte bekomme ich seit kurzen die Fehlermeldung:

War das das nicht zu erkennen. PR welcome.

StefanSchoof, 2022-01-16:
Ja, Fehlerbeschreibung ist nicht immer leicht ;)

PR der die Abhängigkeit hinzufügt oder die composer.lock eincheckt?

andig, 2022-01-16:
Fehlerbeschreibung. Die composer.lock suggeriert nur, dass exakt das die funktionierende Config wäre…