volkszaehler/volkszaehler.org #903: Missing summed value "Verbrauch" for sensortype "universalsensor"

thetruebumblebee, 2022-04-24:
as for sensor type "powersensor" is shown.
I am trying to measure the amount of rain in l/m².

Thx and Kind Regards
tom

J-A-U, 2022-04-25:
Yes. How to compute a summary depends on the physical unit, but the whole point of universal sensor is to give you a channel without unit. With the drawback not be able to calculate consumtion.

For an alternative solution: How does your raw data look like?

maniac-on-moon, 2022-04-25:
Thanks  for your  reply.
The  datasource  is  sending  the last  l/m2 every 300s.

frankrichter, 2022-04-25:
You can use "consumptionsensor".

frankrichter, 2022-04-25:
If your datasource sends l/(m²*s) you need consumptionsensor which
integrates over time to provide consumption values.

If your datasource sends l/m² (delta since last datapoint) you need
universalcounter which only sums up the readings to get consumption values.