volkszaehler/volkszaehler.org #907: composer.lock is not checked in - dependency symfony 6.1 seems broken

wrichter, 2022-05-28:
I have problems rebuilding the docker container (for ARM). The container builds, but a newer version of symfony (6.1) is being pulled in which doesn't work properly. Could you please check in your composer.lock (it is in .gitignore) file so I know which version you're using?

https://getcomposer.org/doc/01-basic-usage.md#commit-your-composer-lock-file-to-version-control

andig, 2022-08-02:
Symfony 6 should not be pulled because that major version is not in the composer.json. If that happens it seems like a composer bug.

andig, 2022-08-02:
> which doesn't work properly.

What happens exactly?

wrichter, 2022-08-02:
https://gist.github.com/wrichter/eb8fa98d50c87ef82898cf6f13b7f6dd

wrichter, 2022-08-02:
In Process.php line 311:

  syntax error, unexpected token ")"

andig, 2022-08-02:
Can you repro without podman? I'm not hitting this locally with php 8.1

andig, 2022-08-02:
And what is `Process.php`?

wrichter, 2022-08-02:
> Can you repro without podman? I'm not hitting this locally with php 8.1. I'm seeing another error with `preg_match` though...

Podman is an alternative to docker. I don't have a local build env, I am using containers only. The problem occurs at run time, not at build time.

> And what is `Process.php`?

Some component in symfony I suppose

andig, 2022-08-02:
Siehe https://github.com/volkszaehler/volkszaehler.org/pull/913#issuecomment-1202420158- kannst Du das mit Apache reproduzieren oder betrifft das nur PPM?

andig, 2022-08-02:
Was mich verwirrt ist, dass hier eine `symfony/process` Version installiert wird, die mit Deinem PHP nicht ausführbar ist. Das dürfte bei sauberen Version Constraints der notwendigen PHP Version eigentlich nicht passieren. Was sagt

    composer why symfony/process

und welche PHP Version hast Du?

andig, 2022-08-02:
@wrichter sollen wir die Diskussion in #913 zusammen führen?

wrichter, 2022-08-02:
Ja gerne - dann können wir das hier schliessen

closed 2022-08-02 by andig