volkszaehler/volkszaehler.org #909: Anzeige im Webbrowser für Impulse falsch

majowi5, 2022-06-17:
Im Web wird jeden Tag um eine Impuls =0,1m3 mehr angezeigt, als tatsächlich verbraucht wurde (=Anzahl der Impulse gemäß MariaDB.)

Im Export für dem gleichen Zeitraum  (von 00:00 bis 00:00 des nächsten Tages) wird auch noch der erste Impuls des nächsten Tages dazugezählt.

```
# source:;volkszaehler.org
# version:;0.3

# uuid:;****
# title:;01 Gas [Bm3]
# from:;2022-06-11 05:35:34
# to:;2022-06-13 05:01:11
# min:;2022-06-13 05:01:11; => ;0.004
# max:;2022-06-12 05:06:18; => ;1.844
# average:;0.017
# consumption:;0.8
# rows:;9
2022-06-12 05:03:03;0.004;1
2022-06-12 05:06:18;1.844;1
2022-06-12 05:09:34;1.838;1
2022-06-12 05:12:52;1.824;1
2022-06-12 05:16:10;1.813;1
2022-06-12 05:19:31;1.795;1
2022-06-12 05:23:09;1.645;1
2022-06-13 05:01:11;0.004
;

![Volkszaehler Gas](https://user-images.githubusercontent.com/83331968/174392598-036a6498-3ed2-4ed7-851d-a4ffc14b0308.png)

J-A-U, 2022-06-17:
Der Impuls wird nicht wirklich "dazugezählt". Es ist vielmehr so das aus den Impulsen der Verbrauch zwischen den Impulsen errechnet wird. Das ist genau die Info was wir für Frontend brauchen um eine Linie zeichnen zu können. Für diese Rechnung braucht man aber halt jeweils den Datenpunkt der vor bzw. nach dem Zeitraum lag.

Impulse sind als Datenquelle einfach schwierig weil man nie weiß was dazwischen genau vor sich gegangen ist. Oder anders ausgedrückt: Impulse enthalten keine Information darüber wann kein Durchfluss mehr stattgefunden hat. wir können das anhand des geringen Verbrauchs nur interpretieren.

Es gibt ein Workaround dazu: "sendzero" im vzlogger. Dann werden nach Ablauf einer festen Zeit 0-Werte in die Datenbank geschrieben. Technisch korrekt ist das allerdings nicht, man erfindet nämlich Daten die der Zähler nie ermittelt hat.

majowi5, 2022-06-18:
Gemäß dem Export wurden 7 Impulse gezählt. 
Die Grafik zeigt aber einen Verbrauch von 0,8 m3 (=8 Impulse); richtig wäre 0,7m3.

[image]

andig, 2022-06-18:
Der erste Impuls wird verschluckt- das ist sozusagen die Nulllinie. Spielt bei längeren Messungen dann keine Rolle mehr.

majowi5, 2022-06-19:
works as designed??

[image]

Es sind 8 Zeilen u. nur 7 sind vom 12.6.

jahir, 2022-06-19:
`rows` sind SQL-Rows, also die Einträge in der Datenbank für den gelieferten Zeitraum. Wie @andig schon geschrieben hat, wird der erste Impuls nicht angezeigt (nur in `from`), dessen Zeitstempel ist aber nötig, um den durchschnittlichen Verbrauch des zweiten Impulses zu berechnen (value / deltaT). 
Der letzte Impuls wird dazugenommen, damit in der Web-UI die Kurve nicht einfach an dem letzten Impuls des angefragten Zeitraums abgehackt wird. Etwas unschön ist dabei tatsächlich, daß die "consumption" davon einfach mitberechnet wird. Besser wäre es evt., den ersten und letzten Impuls zeitanteilsabhängig zu verringern (Dreisatz). Wesentliche Auswirkungen hat das aber nur bei wenigen Impulsen im Zeitfenster, so wie bei dir. `consumption: 0,7` wäre aber m.E nicht korrekt (ohne es ausgerechnet zu haben), da dabei der Anteil zwischen Mitternacht und dem ersten (angezeigten, also zweiten DB-)Impuls und der Anteil des letzten Impulses im angefragten Tag.

Also ja, wurde genau so designed.

majowi5, 2022-06-20:
@jahir
`…Den CSV-Output (oder auch sonstigen) kannst du mit drei Hochkommata ``` einschließen, dann kann man das normal lesen und kopieren,…`

Ich habe diese Darstellung gewählt, weil ich durch den Pfeil und den Text darauf hinweisen wollte, dass zwar im Text **rows: 9** steht, tatsächlich aber nur **8 rows** angezeigt werden.
Da durch Doppelklick auf die Grafik, diese gut leserlich dargestellt wird, habe ich keinen Nachteil gesehen. Teile daraus kopieren kann man aber nicht. Der Inhalt des .csv files steht oben „quoted“ zur Verfügung.

**Zur Thema consumption:**
Derzeit wird auch an Tagen, an denen überhaupt kein Impuls generiert wird, eine consumption von 0,1 m3 angezeigt. 

An folg. Beispiel möchte ich zeigen, warum ich glaube, dass die jetzige Berechnung nicht optimal ist.
Annahme: 
Konstanter Gasfluss von 0,4m3 pro Tag. Ein Impuls pro 0,1m3.
Um 03:00, 09:00, 15:00,21:00 wird ein Impuls generiert. In Summe sind das 4 Impulse.
Im Export würden die 4 Impulse dieses Tages aufscheinen und der Impuls um 03:00 des nächsten Tages. In Summe 5 Impulse.
Im Header würde stehen: 
**Rows: 6
Consumption:5**      (0,5m3)

Die tatsächliche Consumption ist aber 0,4m3. (siehe oben)
Auch bei wesentlich unregelmäßigeren Durchflüssen gleichen sich die Werte vor dem ersten u. nach dem letzen Impuls in etwa aus, sodass man keinen zusätzlichen Impuls rechnen sollte. 

Die Summe der Tageswerte stimmt bei der jetzigen Methode nie mit dem am Gaszähler abgelesenen Werte überein.

jahir, 2022-06-20:
> Derzeit wird auch an Tagen, an denen überhaupt kein Impuls generiert wird, eine consumption von 0,1 m3 angezeigt.

Dass an einem Tag kein Impuls generiert wurde, heißt ja nicht, daß keine consumption statt fand. So wie umgekehrt ein Impuls nicht bedeutet, daß in dieser einen ms 0,1m³ Gas geflossen sind. Ein Impuls bedeutet lediglich, daß seit dem letzten Impuls 0,1m³ Gas verbraucht wurden. Da man die zeitliche Verteilung nicht kennt, nimmt man einfach einen kontinuierlichen Verbrauch an, was i.A. nicht stimmt, aber man weiß es halt auch nicht besser...

Das Thema hatten wir schon vor 10 Jahren oder so. Ja, man kann es besser machen und es ist wahrscheinlich nicht mal sehr aufwendig (hab mal ein bisschen mit dem Code gespielt), aber man muß da schon ein paar Teile anfassen und dabei gut aufpassen, daß nichts kaputt gut. Wenn ich etwas Zeit haben, setzte ich das vielleicht mal um... Bis jetzt hat das aber nicht genug gestört, von daher möchte ich da nichts versprechen. Für deinen Anwendungsfall mag das ungeschickt sein, aber das ist halt nicht so der Normalfall. Meistens hat man ja einen S0-Meter, der ein paar Hundert Impuls pro Tage liefert, da kommt es auf einen mehr oder weniger nicht an. Zumal durch reboot etc. ja auch Impulse verloren gehen können. Von daher kannst du sowieso nicht annehmen, daß die Summe der Impulse genau deinem Verbrauch entspricht. Die Impulse von der middleware pro Tag abzurufen und daraus den Jahresverbrauch zu bestimmen, ist auch eher ein konstruierter Fall. Da würdest du eher das ganze Jahr abrufen und `group=day` oder so benutzen.

majowi5, 2022-06-21:
@jahir

> Dass an einem Tag kein Impuls generiert wurde, heißt ja nicht, daß keine consumption stattfand. So wie umgekehrt ein Impuls nicht bedeutet, daß in dieser einen ms 0,1m³ Gas geflossen sind. Ein Impuls bedeutet lediglich, daß seit dem letzten Impuls 0,1m³ Gas verbraucht wurden.

Da im Sommer das Gas für mehrere Tage ganz abgeschaltet wird, habe ich nun die skurrile Situation, dass an jedem dieser Tage ein Verbrauch von 0,1 m2 als consumption angezeigt wird. Gehe ich von der Tagesansicht auf die Wochenansicht, so wird für die ganze Woche eine consumption von 0,1 m3 angezeigt. Im Grafikteil ist aber nie ein Impuls zu sehen (weil es keine Impulse gibt.)

Bisher ist es mir nicht gelungen einen Vorteil in der realisierten Variante zu sehen, wohl aber einige Nachteile., insbesondere bei einer kleinen Anzahl von Impulsen je Tag. Bei einer größeren Anzahl von Impulsen ist es sowieso egal.

Ich kann aber damit leben, beim Analysieren des Verbrauchs im  GUI  immer 0,1 m3 abzuziehen, bzw. für Analysen die Daten aus MariaDB zu exportieren.

jahir, 2022-06-22:
> Da im Sommer das Gas für mehrere Tage ganz abgeschaltet wird, habe ich nun die skurrile Situation, dass an jedem dieser Tage ein Verbrauch von 0,1 m2 als consumption angezeigt wird. 

Ja, das Problem ist bekannt. Da hatten sich auch schon andere beschwert, daß über Nacht der Verbrauch nicht auf 0 ging, obwohl das Gerät doch ausgeschaltet war. Deswegen wurde ja auch `sendzero` in den vzlogger eingebaut. Wenn man das ganz sauber haben möchte, braucht man halt einen Sensor, der den Durchfluß meldet. Beim Strom machen das neuere Zähler ja oft, bei Gas kenn ich mich leider nicht so aus.

majowi5, 2022-06-28:
Aus mathematischer Sicht finde ich die gewählte Lösung unglücklich, ich kann damit aber leben u. werde das Thema daher nicht weiter verfolgen.

closed 2022-06-28 by majowi5

majowi5, 2022-07-05:
**Zusammenfassung** 

**Die Verbrauchsdaten stimmen am aktuellen Tag bis um Mitternacht!** Dann wird unter allen Umständen 1 Impuls (bei mir 0,1m3) dazugerechnet. 

Vor Mitternacht:
[image]

```
# source:;volkszaehler.org
# version:;0.3

# uuid:;xxx
# title:;01 Gas [Bm3]
# from:;2022-06-29 05:33:31
# to:;2022-06-30 05:30:51
# min:;2022-06-30 05:03:41; => ;0.004
# max:;2022-06-30 05:07:00; => ;1.812
# average:;0.038
# consumption:;0.9
# rows:;10
2022-06-30 05:03:41;0.004;1
2022-06-30 05:07:00;1.812;1
2022-06-30 05:10:19;1.804;1
2022-06-30 05:13:40;1.797;1
2022-06-30 05:17:01;1.785;1
2022-06-30 05:20:24;1.776;1
2022-06-30 05:23:47;1.772;1
2022-06-30 05:27:12;1.756;1
2022-06-30 05:30:51;1.641;1
```

**Nach Mitternacht**

Aus 0,9m3 wird 1m3, obwohl keine Impuls dazugekommen ist.

[image]

```
# source:;volkszaehler.org
# version:;0.3

# uuid:;8b33f740-854a-11ea-bda5-d5a0cf4f9508
# title:;01 Gas [Bm3]
# from:;2022-06-29 05:33:31
# to:;2022-07-01 05:02:12
# min:;2022-07-01 05:02:12; => ;0.004
# max:;2022-06-30 05:07:00; => ;1.812
# average:;0.021
# consumption:;1
# rows:;11
2022-06-30 05:03:41;0.004;1
2022-06-30 05:07:00;1.812;1
2022-06-30 05:10:19;1.804;1
2022-06-30 05:13:40;1.797;1
2022-06-30 05:17:01;1.785;1
2022-06-30 05:20:24;1.776;1
2022-06-30 05:23:47;1.772;1
2022-06-30 05:27:12;1.756;1
2022-06-30 05:30:51;1.641;1
2022-07-01 05:02:12;0.004;1

```