volkszaehler/volkszaehler.org #914: Dockerfile references two different versions of PHP in build and run containers

wrichter, 2022-08-02:
This pulls together the discussion in #907 #908 and #913 . Dependency mismatches create runtime images which fail during execution (see https://gist.github.com/wrichter/eb8fa98d50c87ef82898cf6f13b7f6dd#file-gistfile1-txt-L314-L350 )

https://github.com/volkszaehler/volkszaehler.org/blob/master/Dockerfile#L3 
The Dockerfile builds using `FROM composer:2 AS builder` which currently contains php-8.1.8

```
# podman run -i -t composer php --version
PHP 8.1.8 (cli) (built: Jul 19 2022 03:36:23) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.1.8, Copyright (c) Zend Technologies
```

https://github.com/volkszaehler/volkszaehler.org/blob/master/Dockerfile#L14
The Dockerfile creates a runtime-environment using `FROM php:8.0-alpine` with php-8.0.23

```
# podman run -i -t php:8.0-alpine php --version
PHP 8.0.21 (cli) (built: Jul 19 2022 03:55:25) ( NTS )
Copyright (c) The PHP Group
Zend Engine v4.0.21, Copyright (c) Zend Technologies
```

Suggested Solution: Ensure matching PHP versions are used in build and run env (not sure whether it's recommended to use 8.1 or 8.0).

andig, 2022-08-02:
Happy to take a PR, I'd suggest 8.1.

wrichter, 2022-08-02:
Lovely... the composer docker images happily switch php versions during major and minor versions (e.g. composer:2.0.2 contains php-7.4.11 and 2.0.10 contains php-8.0.2 and 2.1 contains php-8.0.6)... this means if we want to fix without randomly breaking in the future we need to pin it to specific versions.

Anyone have a better idea?

andig, 2022-08-02:
just use `php` and hope for `composer:2` matching the currently best php version?

jahir, 2022-08-02:
maybe we shouldn't use the composer image, it seems to be [not suited for production](https://hub.docker.com/_/composer):
```
PHP version & extensions

Our image is aimed at quickly running Composer without the need for having a PHP runtime installed on your host. You should not rely on the PHP version in our container. We do not provide a Composer image for each supported PHP version because we do not want to encourage using Composer as a base image or a production image.

We try to deliver an image that is as lean as possible, built for running Composer only. Sometimes dependencies or Composer [scripts](https://getcomposer.org/doc/articles/scripts.md) require the availability of certain PHP extensions.

Suggestions:

(optimal) create your own build image and [install](https://getcomposer.org/doc/faqs/how-to-install-composer-programmatically.md) Composer inside it.

Note: Docker 17.05 introduced [multi-stage builds](https://docs.docker.com/develop/develop-images/multistage-build/), simplifying this enormously:

COPY --from=composer /usr/bin/composer /usr/bin/composer

(alternatively) specify the target [platform](https://getcomposer.org/doc/06-config.md#platform) / extension(s) in your composer.json:

{
  "config": {
    "platform": {
      "php": "MAJOR.MINOR.PATCH",
      "ext-something": "MAJOR.MINOR.PATCH"
    }
  }
}
```

wrichter, 2022-08-02:
Thanks @jahir I implemented this approach and it works on my system

StefanSchoof, 2022-08-19:
Können wir das wo jetzt #915 gemerge schlissen?

closed 2022-08-19 by andig