volkszaehler/volkszaehler.org #916: Add/activate Docker integration test

jahir, 2022-08-08:
Wie in #915 bemerkt, laufen aktuell keine Docker-Integrationstests.

@StefanSchoof kannst du das machen oder kurz erläutern, was man dafür machen müßte?

J-A-U, 2022-08-08:
Soweit ich mich erinnere haben wir den mit Absicht weg gelassen um die Rechnerzeit auf Github nicht unnötig aufzublähen. Kostet uns zwar nix, aber das könnte sich ändern wenn wir mit der Resourcen nicht bedacht umgehen.
Siehe dazu die letzten Diskussionen um Tests/Travis/etc.

StefanSchoof, 2022-08-08:
Ich würde vorschlagen als erstes ein einfache Pipeline bauen die das docker image baut und einmal kurz testet, ob es anfragen annimmt. Ich würde das als zusätzlichen build workflow sehen. Dann lässt sich einfacher sehen, ob das ein docker oder andere Fehler ist. Ich könnte einen einfachen Vorschlag erstellen.

Die Frage ist, ob man das nur für den default branch macht oder auch schon für PRs und/oder zeitlich wiederholend. Docker hat auf github einen ganz guten Cache der nach meiner Erfahrung viele builds deutlich abkürzt. Aber es benötigt ohne Frage Ressourcen auf dem github agents. Wenn es aber Fehler findet, spart es menschliche Ressourcen.

Geht es hier nur um einen docker integration test oder auch um das aktualisieren des images auf docker hub und/oder github container registry?

jahir, 2022-08-10:
> Ich würde vorschlagen als erstes ein einfache Pipeline bauen die das docker image baut und einmal kurz testet, ob es anfragen annimmt. Ich würde das als zusätzlichen build workflow sehen. Dann lässt [...]

Genau sowas hatte ich mir vorgestellt. Wäre super, wenn du das machen könntest.

> Die Frage ist, ob man das nur für den default branch macht oder auch schon für PRs und/oder zeitlich wiederholend.

Wenn die Rechenzeit in der Vergangenheit ein Problem war, müssen wir wohl schon etwas aufpassen, könnten aber aber im Problemfall ja auch zurückgedreht werden.
Bei PRs wäre m.E. der größte Nutzen (und so viele haben wir i.A. ja auch nicht).

Das Image bei Docker Hub wäre schon schön, das Image dort ist ja schon zwei Jahre alt. @andig, du hast das zuletzt aktualisiert, was meinst du dazu?

andig, 2022-08-10:
Rechenzeit ist bei GH kein Problem mehr, build-on-commit auf master sollte völlig genügen. Im Moment liegt das Image in einer Docker Org- dafür bräuchten wir einen Maintainer der das mit mehr Elan als ich voran treiben könnte, gerne auch für die Automatisierung des Repos hier. Den Travis Umzug hab ich ja schon nicht mehr voll umgesetzt.... Findet sich da jemand?

StefanSchoof, 2022-08-10:
Ich kann mir vorstellen für die docker images ein paar PR zu machen und ein Auge drauf zu haben. Ich bräuchte, aber Leute die kritische auf die PRs schauen und mitdenken, in welche Richtung das gehen soll. 

@andig Was meinst du mit "Automatisierung des Repos"?

PR #917 (2022-08-13): Dockerbuild
Ein Vorschlag für einen docker build und push:
- pushed in den docker hub und in die github container registry
- Läuft bei jedem push auf master und aktualisiert den latest tag
- Lässt sich manual anstoßen und pushed dann mit einen sha commit id als tag
- testet das image einmal, ob es startet und ein bisschen ob anfragen funktionieren

Benötigt noch DOCKERHUB_USERNAME und DOCKERHUB_TOKEN als secrets in den actions

Resolves #916
closed 2022-08-13 by andig