volkszaehler/volkszaehler.org #923: Virtualsensor with hourly Price (e.g. Tibber)

Thmas47051, 2023-01-26:
How to configure a virtual sensor to calculate the costs in a specific time? Like Power Consumption does.
I made a virtual sensor with val(in1)*val(in2) with in1=Power and in2=Price per kWh
The total consumtion (total price expected) is not calculated as expected.
any hints?
Example: https://www.photovoltaikforum.com/thread/191811-verst%C3%A4ndnisfrage-dynamischer-strompreis/?postID=2956600#post2956600
in VirtualInterpreter.php/generateData evaluates the formula, multiplied by the time, so i would expext the cumulated cost.
An hints?

frankrichter, 2023-01-26:
Loggst du den Preis nur 1 x pro Stunde? Wahrscheinlich berechnet die
Middleware dann mindestens ein Intervall von 1 h bzw. Vielfache davon.

Thmas47051, 2023-01-27:
habe für heute minütlich die Preise gepusht - keine Änderung.
Merkwürdig: wenn ich als Formel nur val(in1) eingebe (aktuelle Leistung Haus), würde ich erwarten, dass der virtuelle Kanal exakt dem Kanal folgt. Es kommt aber nur Unsinn heraus.