volkszaehler/volkszaehler.org #928: s0 zähler mit ardurino über usb auslesen

yukimaotec, 2023-03-25:
hallo 

Ich weiss was für ein trottel will mit einen ardurino einen zähler auslesen leiber nen teuren raspi nehmen aber ich bin zurzeit nicht bereit knap 100€ für ne pi auszugeben daher meine idee einfach ne uno über usb an den vz server direkt zu verwenden. leider finde ich nicht derkleichen online , ich habe auch nicht die expertise um mir da etwas selber zuschreiben möchte aber auch nicht eine fertig lösung kaufen mussen 

meine bitte wen ich sie stellen kann, ob mir jemand wenigstens die richtung weisen kann hab seit zwei wochen den zähler vom elektriker vewrbaut bekommen und seit den keine möglichkeit ihn auszulehsen, hoffe das dass keine tote nummer wird und ich am ende doch nen proprietäre lösung nutzen muss

mit freundlichen grüssen

J-A-U, 2023-03-25:
Das ist git mehr der Ort für konkrete Probleme mit dem Code. Anwendungsfragen haben wir lieber auf der Mailingliste oder alternativ im Photovoltaikforum.

Zu deiner Frage:
https://wiki.volkszaehler.org/hardware/controllers/espeasy
https://wiki.volkszaehler.org/hardware/controllers/esp8266-ir-lesekopf
https://www.ebay.de/itm/314321775948
Zum Hichi gibt es im PV-Forum einen seeehr langen Thread.

yukimaotec, 2023-03-25:
erstens danke blicke nicht gans durch was eure mailing liste angeht 
zweitens hab ich nur grad nen uno da und da ist mein problem ich möchte das alles auf einen pc mit uno betreiben und auch nur mit einen zähler nicht an zwei stellen und auch nicht über ethernet, klar das frontend soll über netzwerk verfügbar sein mehr auch nicht und da finde ich nichts wie sag ich den vzlogger das er über usb daten nehemn soll wo finde ich nen beispielcode inden ich benutzen könnte. ich will ichts komplexes nur wieviel strom meine wohnung verbraucht und das in en grafen mehr nicht

J-A-U, 2023-03-25:
> wie sag ich den vzlogger das er über usb daten nehemn soll 

Der (Standard-) Lesekopf den die meisten User am Rpi hängen haben wird per USB angebunden.
Aktuelle Empfehlung ist da (auch) das entsprechende Modell von Hichi.

> wo finde ich nen beispielcode inden ich benutzen könnte.

Das Wiki ist voll von Beispielkonfigurationen! 
https://wiki.volkszaehler.org/hardware/channels/meters/power/edl-ehz

Und der vzlogger lässt sich auch auf PC (mit Linux natürlich) kompilieren, funktioniert da genauso gut. Support _dazu_ ist _hier_ aber in doppelter Hinsicht fehl am Platz.

r00t-, 2023-03-28:
S0 mit einem AVR auszulesen ist eine der aeltesten varianten des volkszaehlers: https://wiki.volkszaehler.org/software/controller/ethersex
zum direkt anschliessen an einen PC reicht ein rs232-port: https://wiki.volkszaehler.org/hardware/controllers/s0-an-rs232
(J-A-U hat angenommen, dass du einen elektronischen zaehler auslesen willst, aber du erwaehnst nur S0.)

closing as off-topic