volkszaehler/volkszaehler.org #932: aggregate fails due to division by zero

mh-er, 2023-04-25:
Aggregation for level 1 (minute) fails due to division by zero.
The problem is caused in this part of the SQL statement that is composed by Aggregate.php:
     COALESCE( SUM(agg.val_by_time) / (MAX(agg.timestamp) - MIN(agg.prev_timestamp)),AVG(agg.value)) AS value

**Reason:** MAX(agg.timestamp) = MIN(agg.prev_timestamp))
This happens when 
- the first data value to be used for aggregation has a "minute" timestamp (e.g. 1.672.668.600.000 which is 2023-01-02 15:10:00)  
- and there is only this value for this minute slot (i.e. the next value has a timestamp >= 2023-01-02 15:11:00)

**Proposed solution:** change lib/Util/Aggregate.php lines 257-262:

Current code:
				$intialTimestamp =
					'UNIX_TIMESTAMP(DATE_ADD(' .
					'FROM_UNIXTIME(MAX(timestamp) / 1000, ' . $format . '), ' .
					'INTERVAL 1 ' . $level . ')) * 1000 ' .
					'FROM aggregate ' .
					'WHERE type = ? AND aggregate.channel_id = ?';
New code:
				$intialTimestamp =
					'UNIX_TIMESTAMP(DATE_ADD(' .
					'FROM_UNIXTIME(MAX(timestamp) / 1000,  "%Y-%m-%d %H:%i:%s"), ' .
					'INTERVAL 0 ' . $level . ')) * 1000 ' .
					'FROM aggregate ' .
					'WHERE type = ? AND aggregate.channel_id = ?';
**Explanation:** with the new code, the initial @prev_timestamp is set to the timestamp of the last entry in the aggregate table which is always smaller than the next bigger data timestamp. (actually, this is also a little more accurate for the time weighting of the value compared to the current implemetation).
Please note both the new date format and the new INTERVAL 0 .

Remark 1: the problem will occur for all aggregation levels, although the probability that there is only one data value in the time slot (minute, hour, day, year) becomes smaller the longer the slot is.
Remark 2: Therefore, it is possible to "hard code" the date format independent from the aggregation level because it will provide a solution for all levels.

Attached an example of the problem that I found in my database:
data5_c.jpg: screenshot of data table
agg5.jpg: screenshot of aggregate table with timestamp where problem occured
agg5_corrected.jpg: screenshot of aggregate table with corrected sql command
aggMinuteCorrected.md: corrected SQL command used to get agg5_corrected.jpg
 

[image]
[image]
[image]
[aggMinuteCorrected.md](https://github.com/volkszaehler/volkszaehler.org/files/11325158/aggMinuteCorrected.md)

andig, 2023-04-26:
Excellent analysis. Would you be able to open a PR?

mh-er, 2023-04-27:
I implemented the solution and put a PR.

PR #933 (2023-05-04): Fix  aggregate fails due to division by zero #932
Fix #932.

The cron job with the changed Aggregate.php is updating the aggregate table for minute and hour in my volkszaehler environment (day update has not run yet).
closed 2023-05-04 by justinotherguy