volkszaehler/volkszaehler.org #938: Network Error - Bad Gateway after Git update

dm82m, 2023-05-22:
Hey guys,

I get a Network Error of a virtual sensor after updating Git repository:
[image]

After updating from c959c701..29caa311:
[image]

Any advice?

Best,
Dirk

dm82m, 2023-05-22:
@jahir I guess the related commit is this one: https://github.com/volkszaehler/volkszaehler.org/commit/3bf0eac68a14157f9dba88b97f4f591389eb9d98

J-A-U, 2023-05-22:
Forgot to run composer?

J-A-U, 2023-05-22:
I suggest to take a look in the php error_log.

dm82m, 2023-05-22:
```
May 22 10:47:35 raspberrypi php[30697]: An exception was thrown by the bridge. Forcing restart of the worker. The exception was: TypeError: Return value of Volkszaehler\Interpreter\Virtual\TimestampIterator::current() must be an instance of Volkszaehler\Interpreter\Virtual\mixed, float returned in /home/pi/volkszaehler.org/lib/Interpreter/Virtual/TimestampItera>
May 22 10:47:35 raspberrypi php[30697]: Stack trace:
May 22 10:47:35 raspberrypi php[30697]: #0 /home/pi/volkszaehler.org/lib/Interpreter/Virtual/TimestampGenerator.php(50): Volkszaehler\Interpreter\Virtual\TimestampIterator->current()
May 22 10:47:35 raspberrypi php[30697]: #1 [internal function]: Volkszaehler\Interpreter\Virtual\TimestampGenerator->Volkszaehler\Interpreter\Virtual\{closure}()
May 22 10:47:35 raspberrypi php[30697]: #2 /home/pi/volkszaehler.org/lib/Interpreter/Virtual/TimestampGenerator.php(54): array_reduce()
May 22 10:47:35 raspberrypi php[30697]: #3 /home/pi/volkszaehler.org/lib/Interpreter/VirtualInterpreter.php(214): Volkszaehler\Interpreter\Virtual\TimestampGenerator->getIterator()
May 22 10:47:35 raspberrypi php[30697]: #4 /home/pi/volkszaehler.org/lib/Interpreter/Interpreter.php(160): Volkszaehler\Interpreter\VirtualInterpreter->generateData()
May 22 10:47:35 raspberrypi php[30697]: #5 /home/pi/volkszaehler.org/lib/View/JSON.php(215): Volkszaehler\Interpreter\Interpreter->getIterator()
May 22 10:47:35 raspberrypi php[30697]: #6 /home/pi/volkszaehler.org/lib/View/JSON.php(149): Volkszaehler\View\JSON->renderInterpreter()
May 22 10:47:35 raspberrypi php[30697]: #7 /home/pi/volkszaehler.org/lib/View/JSON.php(91): Volkszaehler\View\JSON->renderDeferred()
May 22 10:47:35 raspberrypi php[30697]: #8 /home/pi/volkszaehler.org/vendor/symfony/http-foundation/StreamedResponse.php(109): Volkszaehler\View\JSON->Volkszaehler\View\{closure}()
May 22 10:47:35 raspberrypi php[30697]: #9 /home/pi/volkszaehler.org/vendor/php-pm/httpkernel-adapter/src/Bridges/HttpKernel.php(301): Symfony\Component\HttpFoundation\StreamedResponse->sendContent()
May 22 10:47:35 raspberrypi php[30697]: #10 /home/pi/volkszaehler.org/vendor/php-pm/httpkernel-adapter/src/Bridges/HttpKernel.php(90): PHPPM\Bridges\HttpKernel->mapResponse()
May 22 10:47:35 raspberrypi php[30697]: #11 /home/pi/volkszaehler.org/vendor/php-pm/php-pm/src/ProcessSlave.php(454): PHPPM\Bridges\HttpKernel->handle()
May 22 10:47:35 raspberrypi php[30697]: #12 /home/pi/volkszaehler.org/vendor/php-pm/php-pm/src/ProcessSlave.php(418): PHPPM\ProcessSlave->handleRequest()
May 22 10:47:35 raspberrypi php[30697]: #13 /home/pi/volkszaehler.org/vendor/react/http/src/Io/MiddlewareRunner.php(49): PHPPM\ProcessSlave->onRequest()
May 22 10:47:35 raspberrypi php[30697]: #14 /home/pi/volkszaehler.org/vendor/react/http/src/Io/MiddlewareRunner.php(54): React\Http\Io\MiddlewareRunner->call()
May 22 10:47:35 raspberrypi php[30697]: #15 /home/pi/volkszaehler.org/vendor/react/http/src/Middleware/RequestBodyParserMiddleware.php(34): React\Http\Io\MiddlewareRunner->React\Http\Io\{closure}()
May 22 10:47:35 raspberrypi php[30697]: #16 /home/pi/volkszaehler.org/vendor/react/http/src/Io/MiddlewareRunner.php(59): React\Http\Middleware\RequestBodyParserMiddleware->__invoke()
May 22 10:47:35 raspberrypi php[30697]: #17 /home/pi/volkszaehler.org/vendor/react/http/src/Io/MiddlewareRunner.php(54): React\Http\Io\MiddlewareRunner->call()
May 22 10:47:35 raspberrypi php[30697]: #18 /home/pi/volkszaehler.org/vendor/react/http/src/Middleware/RequestBodyBufferMiddleware.php(44): React\Http\Io\MiddlewareRunner->React\Http\Io\{closure}()
May 22 10:47:35 raspberrypi php[30697]: #19 /home/pi/volkszaehler.org/vendor/react/http/src/Io/MiddlewareRunner.php(59): React\Http\Middleware\RequestBodyBufferMiddleware->__invoke()
May 22 10:47:35 raspberrypi php[30697]: #20 /home/pi/volkszaehler.org/vendor/react/http/src/Io/MiddlewareRunner.php(40): React\Http\Io\MiddlewareRunner->call()
May 22 10:47:35 raspberrypi php[30697]: #21 /home/pi/volkszaehler.org/vendor/react/http/src/Io/StreamingServer.php(150): React\Http\Io\MiddlewareRunner->__invoke()
May 22 10:47:35 raspberrypi php[30697]: #22 /home/pi/volkszaehler.org/vendor/react/http/src/Io/StreamingServer.php(114): React\Http\Io\StreamingServer->handleRequest()
May 22 10:47:35 raspberrypi php[30697]: #23 /home/pi/volkszaehler.org/vendor/evenement/evenement/src/Evenement/EventEmitterTrait.php(123): React\Http\Io\StreamingServer->React\Http\Io\{closure}()
May 22 10:47:35 raspberrypi php[30697]: #24 /home/pi/volkszaehler.org/vendor/react/http/src/Io/RequestHeaderParser.php(108): Evenement\EventEmitter->emit()
May 22 10:47:35 raspberrypi php[30697]: #25 /home/pi/volkszaehler.org/vendor/evenement/evenement/src/Evenement/EventEmitterTrait.php(123): React\Http\Io\RequestHeaderParser->React\Http\Io\{closure}()
May 22 10:47:35 raspberrypi php[30697]: #26 /home/pi/volkszaehler.org/vendor/react/stream/src/Util.php(71): Evenement\EventEmitter->emit()
May 22 10:47:35 raspberrypi php[30697]: #27 /home/pi/volkszaehler.org/vendor/evenement/evenement/src/Evenement/EventEmitterTrait.php(123): React\Stream\Util::React\Stream\{closure}()
May 22 10:47:35 raspberrypi php[30697]: #28 /home/pi/volkszaehler.org/vendor/react/stream/src/DuplexResourceStream.php(196): Evenement\EventEmitter->emit()
May 22 10:47:35 raspberrypi php[30697]: #29 /home/pi/volkszaehler.org/vendor/react/event-loop/src/StreamSelectLoop.php(246): React\Stream\DuplexResourceStream->handleData()
May 22 10:47:35 raspberrypi php[30697]: #30 /home/pi/volkszaehler.org/vendor/react/event-loop/src/StreamSelectLoop.php(213): React\EventLoop\StreamSelectLoop->waitForStreamActivity()
May 22 10:47:35 raspberrypi php[30697]: #31 /home/pi/volkszaehler.org/vendor/php-pm/php-pm/src/ProcessSlave.php(377): React\EventLoop\StreamSelectLoop->run()
May 22 10:47:35 raspberrypi php[30697]: #32 /tmp/dbgvpyInn(37): PHPPM\ProcessSlave->run()
May 22 10:47:35 raspberrypi php[30697]: #33 {main}
```

jahir, 2023-05-22:
argh... mixed is only supported in php 8+, so you probably are using something older. Looking into it... timestamps should be int, but we sometimes use float (to minimize rounding errors in calculations).

dm82m, 2023-05-22:
I am still on Wheezy as Bookworm is not released yet.

jahir, 2023-05-22:
could you try #939 (or checkout branch timestamp_iterators_return_type)?

I really hope you mean, bullseye, because wheezy is [really old](https://wiki.debian.org/PHP#Available_versions) (even for debian standards)

dm82m, 2023-05-22:
branch _timestamp_iterators_return_type_ works

closed 2023-05-22 by jahir

dm82m, 2023-05-22:
Thanks for the quick fix!

PR #939 (2023-05-22): use ReturnTypeWillChange attribute instead of mixed return types
mixed is only supported in php 8+

should fix #938