volkszaehler/volkszaehler.org #943: Service Temporarily Unavailable

dm82m, 2023-07-25:
After `sudo composer update` and restart of middleware the browser shows `Service Temporarily Unavailable`.

composer update:
```
pi@raspberrypi:~/volkszaehler.org $ sudo composer update
Do not run Composer as root/super user! See https://getcomposer.org/root for details
Loading composer repositories with package information
Warning from https://repo.packagist.org: Support for Composer 1 is deprecated and some packages will not be available. You should upgrade to Composer 2. See https://blog.packagist.com/deprecating-composer-1-support/
Info from https://repo.packagist.org: #StandWithUkraine
Updating dependencies (including require-dev)
Package operations: 0 installs, 4 updates, 0 removals
  - Updating react/promise (v2.10.0 => v3.0.0): Loading from cache
  - Downgrading php-pm/php-pm (2.4.0 => 2.1.3): Loading from cache
  - Updating phpunit/phpunit (9.6.9 => 9.6.10): Loading from cache
  - Updating phpstan/phpstan (1.10.24 => 1.10.26): Loading from cache
Package influxdb/influxdb-php is abandoned, you should avoid using it. No replacement was suggested.
Writing lock file
Generating autoload files
composer/package-versions-deprecated: Generating version class...
composer/package-versions-deprecated: ...done generating version class
```

Syslog:
```
Jul 25 12:17:01 raspberrypi CRON[9068]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Jul 25 12:17:32 raspberrypi systemd[1]: Stopping Volkszaehler.org Middleware...
Jul 25 12:17:32 raspberrypi php[4435]: Server is shutting down.
Jul 25 12:17:32 raspberrypi php[4435]: Stopping the process manager.
Jul 25 12:17:33 raspberrypi systemd[1]: middleware.service: Succeeded.
Jul 25 12:17:33 raspberrypi systemd[1]: Stopped Volkszaehler.org Middleware.
Jul 25 12:17:33 raspberrypi systemd[1]: middleware.service: Consumed 16min 36.069s CPU time.
Jul 25 12:17:33 raspberrypi systemd[1]: Started Volkszaehler.org Middleware.
Jul 25 12:17:33 raspberrypi php[9116]: Read configuration /home/pi/volkszaehler.org/etc/middleware.json, modified by command arguments.
Jul 25 12:17:33 raspberrypi php[9116]: /
Jul 25 12:17:33 raspberrypi php[9116]: +---------------------------+-----------------------------------------+
Jul 25 12:17:33 raspberrypi php[9116]: | host                      | 0.0.0.0                                 |
Jul 25 12:17:33 raspberrypi php[9116]: | port                      | 8080                                    |
Jul 25 12:17:33 raspberrypi php[9116]: | workers                   | 8                                       |
Jul 25 12:17:33 raspberrypi php[9116]: | static-directory          | /var/www/volkszaehler.org/htdocs        |
Jul 25 12:17:33 raspberrypi php[9116]: | logging                   | 0                                       |
Jul 25 12:17:33 raspberrypi php[9116]: | max-requests              | 1000                                    |
Jul 25 12:17:33 raspberrypi php[9116]: | bridge                    | HttpKernel                              |
Jul 25 12:17:33 raspberrypi php[9116]: | bootstrap                 | Volkszaehler\Server\PPMBootstrapAdapter |
Jul 25 12:17:33 raspberrypi php[9116]: | app-env                   | dev                                     |
Jul 25 12:17:33 raspberrypi php[9116]: | debug                     | 0                                       |
Jul 25 12:17:33 raspberrypi php[9116]: | max-execution-time        | 30                                      |
Jul 25 12:17:33 raspberrypi php[9116]: | memory-limit              | -1                                      |
Jul 25 12:17:33 raspberrypi php[9116]: | limit-concurrent-requests |                                         |
Jul 25 12:17:33 raspberrypi php[9116]: | request-body-buffer       |                                         |
Jul 25 12:17:33 raspberrypi php[9116]: | ttl                       | 0                                       |
Jul 25 12:17:33 raspberrypi php[9116]: | populate-server-var       | 1                                       |
Jul 25 12:17:33 raspberrypi php[9116]: | socket-path               | .ppm/run/                               |
Jul 25 12:17:33 raspberrypi php[9116]: | pidfile                   | .ppm/ppm.pid                            |
Jul 25 12:17:33 raspberrypi php[9116]: | reload-timeout            | 30                                      |
Jul 25 12:17:33 raspberrypi php[9116]: | cgi-path                  | /usr/bin/php                            |
Jul 25 12:17:33 raspberrypi php[9116]: +---------------------------+-----------------------------------------+
Jul 25 12:17:33 raspberrypi php[9116]: Starting PHP-PM with 8 workers, using StreamSelectLoop ...
Jul 25 12:17:34 raspberrypi php[9116]: --- Worker 5507 stderr ---
Jul 25 12:17:34 raspberrypi php[9116]: 2023-07-25T12:17:34+02:00 [critical] Uncaught Error: Call to undefined method React\Promise\Internal\FulfilledPromise::done()
Jul 25 12:17:34 raspberrypi php[9116]: Worker permanently closed during PHP-PM bootstrap. Not so cool. Not your fault, please create a ticket at github.com/php-pm/php-pm with the output of `ppm start -vv`.
Jul 25 12:17:34 raspberrypi php[9116]: --- Worker 5503 stderr ---
Jul 25 12:17:34 raspberrypi php[9116]: 2023-07-25T12:17:34+02:00 [critical] Uncaught Error: Call to undefined method React\Promise\Internal\FulfilledPromise::done()
Jul 25 12:17:34 raspberrypi php[9116]: Worker permanently closed during PHP-PM bootstrap. Not so cool. Not your fault, please create a ticket at github.com/php-pm/php-pm with the output of `ppm start -vv`.
Jul 25 12:17:34 raspberrypi php[9116]: --- Worker 5506 stderr ---
Jul 25 12:17:34 raspberrypi php[9116]: 2023-07-25T12:17:34+02:00 [critical] Uncaught Error: Call to undefined method React\Promise\Internal\FulfilledPromise::done()
Jul 25 12:17:34 raspberrypi php[9116]: Worker permanently closed during PHP-PM bootstrap. Not so cool. Not your fault, please create a ticket at github.com/php-pm/php-pm with the output of `ppm start -vv`.
Jul 25 12:17:34 raspberrypi php[9116]: --- Worker 5501 stderr ---
Jul 25 12:17:34 raspberrypi php[9116]: 2023-07-25T12:17:34+02:00 [critical] Uncaught Error: Call to undefined method React\Promise\Internal\FulfilledPromise::done()
Jul 25 12:17:34 raspberrypi php[9116]: Worker permanently closed during PHP-PM bootstrap. Not so cool. Not your fault, please create a ticket at github.com/php-pm/php-pm with the output of `ppm start -vv`.
Jul 25 12:17:34 raspberrypi php[9116]: --- Worker 5505 stderr ---
[... 20 lines cut ...]
Jul 25 12:18:39 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:18:46 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:18:57 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:19:08 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:19:19 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:19:20 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:19:30 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:19:41 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:19:52 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:20:01 raspberrypi CRON[9199]: (pi) CMD (php /var/www/volkszaehler.org/bin/aggregate run -m delta -l minute >/dev/null)
Jul 25 12:20:03 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:20:14 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:20:25 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:20:31 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:20:36 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:20:47 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:20:58 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
[... cut ...]
Jul 25 12:21:31 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:21:42 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:21:42 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:21:53 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
Jul 25 12:22:04 raspberrypi php[9116]: No slaves available to handle the request and timeout 10 seconds exceeded
```

jahir, 2023-09-04:
> Warning from https://repo.packagist.org: Support for Composer 1 is deprecated and some packages will not be available. You should upgrade to Composer 2. See https://blog.packagist.com/deprecating-composer-1-support/

Do you really have a composer version 1.x?? current is 2.6.2. please run `composer self-update` or get the latest one from https://getcomposer.org/. If this doesn't help, please show your composer.json and .lock. The downgrade of php-pm looks suspicious (as the upgrade of react/promise to the next major version).

dm82m, 2023-09-04:
thanks for your help, issue is solved!

closed 2023-09-04 by dm82m