volkszaehler/volkszaehler.org #949: PHP 8.2: Dynamic Properties are deprecated

evoracer, 2023-09-19:
After an Upgrade from Debian Bullseye (PHP 7.4) to Debian Bookworm (PHP 8.2) I got a lot of deprecation Warnings in the Apache error Log /var/log/apache2/error.log.

`[info] Deprecated: Creation of dynamic property Webpatser\\Uuid\\Uuid::$bytes is deprecated`
`[info] Deprecated: Creation of dynamic property Webpatser\\Uuid\\Uuid::$string is deprecated`
`[info] Deprecated: Creation of dynamic property Webpatser\\Uuid\\Uuid::$uuid_ordered is deprecated`

Everytime i add some Values to volkszaehler via curl i get those 3 Warnings in the Apache error.log

`curl -s  "http://$VZHOST/middleware.php/data/<CHANNEL UUID>.json?operation=add&value=$VALUE"`

So my error log is about 100M per Day full of these deprecation Warnings.

jahir, 2023-09-20:
did you run `composer update`? what's in your composer.json?

evoracer, 2023-09-21:
yes
```
$ composer update
Loading composer repositories with package information
Info from https://repo.packagist.org: #StandWithUkraine
Updating dependencies
Nothing to modify in lock file
Installing dependencies from lock file (including require-dev)
Nothing to install, update or remove
Package influxdb/influxdb-php is abandoned, you should avoid using it. No replacement was suggested.
Generating autoload files
79 packages you are using are looking for funding.
Use the `composer fund` command to find out more!
No security vulnerability advisories found
```

and my composer.json

```
{
    "name": "volkszaehler/volkszaehler",
    "type": "project",
    "description": "volkszaehler is a free smart meter implementation with focus on data privacy",
    "homepage": "https://www.volkszaehler.org/",
    "license": "GPL-3.0",
    "support": {
        "wiki": "https://wiki.volkszaehler.org/",
        "email": "volkszaehler@lists.volkszaehler.org"
    },
    "require": {
        "php": "^7.3|^8.0",
        "doctrine/orm": "^2.9",
        "doctrine/dbal": "^2.12",
        "doctrine/annotations": "^1.13.2",
        "andig/dbcopy": "^1.4",
        "andig/php-shunting-yard": "^1.0",
        "symfony/cache": "^5.0",
        "symfony/console": "^4.4|^5.0",
        "symfony/http-kernel": "^4.4|^5.0",
        "symfony/http-foundation": "^4.4|^5.0",
        "symfony/routing": "^4.4|^5.0",
        "symfony/yaml": "^4.4|^5.0",
        "cboden/ratchet": "^0.4",
        "ringcentral/psr7": "^1.2",
        "php-pm/php-pm": "^2.0",
        "php-pm/httpkernel-adapter": "^2.0",
        "webpatser/laravel-uuid": "^4.0"
    },
    "require-dev": {
        "phpunit/phpunit": "^9.5",
        "nyholm/psr7": "^1.1",
        "symfony/psr-http-message-bridge": "^2.0",
        "guzzlehttp/guzzle": "^7.0",
        "jdorn/sql-formatter": "^1.2",
        "friendsofphp/php-cs-fixer": "^3.0",
        "phpstan/phpstan": "^1.4.8"
    },
    "autoload": {
        "psr-4": {
            "Volkszaehler\\": "lib/",
            "Tests\\": "test/"
        }
    }
}
```

jahir, 2023-09-21:
try this patch and run `composer update` again

```
--- a/composer.json
+++ b/composer.json
@@ -25,7 +25,7 @@
         "ringcentral/psr7": "^1.2",
         "php-pm/php-pm": "^2.0",
         "php-pm/httpkernel-adapter": "^2.0",
-        "webpatser/laravel-uuid": "^4.0"
+        "webpatser/laravel-uuid": "dev-master#f82715c6bc6a4fcc61bb020cb4fddeb8cc73bbea"
     },
     "require-dev": {
         "phpunit/phpunit": "^9.5",
```

evoracer, 2023-09-21:
This works. Thank you!

jahir, 2023-09-21:
fixed in 7770dc8b614a0e1c8bb30c38a1a6878a961d905b

closed 2023-09-21 by jahir