volkszaehler/volkszaehler.org #961: Is this project still alive?

loxy, 2025-01-08:
Because there was no commit since two years and many open issues and PR.

Just asking if it is worth to jump in here. Or do you know guys similar projects?

justinotherguy, 2025-01-08:
Hi @loxy,
well - although I agree, the project could be more active, I would still declare it as alive.
Apparently it seems to do the job for most of the people using it.
Question is, what you're looking for.
Regards, J.

loxy, 2025-01-08:
Thank you for the answer. Looking for nothing special. Just asking. It is always a bad sign when last commit is years ago. At least, I would expect some depency updates when features are complete

r00t-, 2025-01-08:
i think nowadays many people use various integrations/stacks (homeassistant, mqtt stuff, influxdb/grafana) instead of the vz.org middleware.
not to say that it doesn't work anymore. maybe it's a bit too clunky/specific/niche.
i personally use vzlogger with influxdb/grafana. and have nothing that calculates consumption from power.

loxy, 2025-01-09:
Thank you for your insights! Yes, I guess I go with [this solution](https://github.com/flohoff/easymeterq3d-topcase) and using MQTT with Grafana.

closed 2025-01-09 by loxy